Ultimaker/Cura #8966: 4.8 (win 10) Gui does not scale properly when on a monitor with  a different resolution and the main monitor

Kilill, 2020-12-18:
**Application version**
4.8
**Platform**
Win 10

**Printer**
Wanaho Duplicator i3 Plus

**Reproduction steps**

2 Displays, main: 4k (3849 x 2160) , Scaling 150% , Secondary : 1920x1200 Scaling 100%

- Start cura, observ that the initial load window is located on the main screen , 
- move cura to secondary screen, Gui will now still be scaled up 150%
- Minimize and close Cura, this should, according to MS lock the main gui to the screen for the next start
- Start cura again
- Load window is still on main screen in correct scale (150%)
- Main gui window opens up on secondary screen , still 150% incorrectly scaled.

**Actual results**
Gui  is incorrectly scaled on secondary display ie enlarged too 150%

**Expected results**
Gui should pick up scaling from the actual monitor

**Additional information**
Qt is supposedly DPI aware per monitor, but it does not seem that this is propagated to cura. ( or perhaps Uranium ?)

Ghostkeeper, 2020-12-24:
Apparently my colleagues have seen this before, but I couldn't find this immediately.

We're planning an upgrade to Qt 5.15. Hopefully that fixes it for us. It was decided this morning that if this bug persists after the upgrade we'll look into fixing this issue ourselves.

jhwblender, 2021-06-11:
I am having this issue too: I start Cura and it will always scale to 150% with my main 4k monitor even though I move it to and start it in a different screen with 100% scaling.

Ghostkeeper, 2021-06-21:
@jhwblender Could you try this with Cura 4.9?

jhwblender, 2021-06-21:
I tried it with Cura 4.9, I still get the same issue. The picture shown is my entire 1080p monitor.
[image]

jhwblender, 2021-09-14:
Using the new Cura 4.11 and the issue still persists. @Ghostkeeper Do you have any updates on this?

Ghostkeeper, 2021-09-21:
We've had this issue on our backlog since June (developers, see CURA-8352). Somewhere in July we made an attempt to fix this, using [this suggestion](https://stackoverflow.com/questions/35816944/improper-scaling-of-qt-desktop-application-in-windows-10/36057995#36057995) and [a guide from Qt](https://doc.qt.io/qt-5/highdpi.html) but without much success. It now appears like we may have to make a signal to trigger all interface elements to update or something, or perhaps implement this through the theme somehow. But it's a difficult issue altogether.

The way we understand the issue now is that upon start-up Cura will start with the scale of the monitor Cura starts on, but if you drag the screen around during runtime it doesn't adapt to the changing pixel density.

brdenos, 2022-04-06:
Issue persists on 4.13.1 . 4k laptop screen (main) okay, 1920x1080 secondary monitor all icons and text oversized. Even if I move the program to second screen, close from there, and re-open on second screen. I couldn't find a combination of per-app settings in windows to address DPI issues that would resolve this.

fvrmr, 2022-04-26:
With Cura 5.0 beta we can not reproduce this issue anymore, this is probably because we upgraded to Qt6! 
I will close this issue, but please let us know if it isn't fixed for you!

closed 2022-04-26 by fvrmr

brdenos, 2022-04-27:
Overall, the scaling is happening now on all elements! BUT.... some elements are not positioning correctly. There is an extra black bar at the top of my screen between the window bar and the top of the main Cura content, and all the icon "click points" and entry fields are offset by the height of that bar. So to close the notifications like "this plugin is not working..." I had to click about 5mm above the "X". Could just be a unique scaling issue for my computer graphics though. 
[image]