Ultimaker/Cura #8967: ChangeAtZ 5.2.1 output text on LCD too long

muenznej, 2020-12-18:
**Application version**
4.7.0

**Platform**
System:    Kernel: 5.4.0-58-generic x86_64 bits: 64 compiler: gcc v: 9.3.0 Desktop: Cinnamon 4.6.7 
           wm: muffin dm: LightDM Distro: Linux Mint 20 Ulyana base: Ubuntu 20.04 focal 
**Printer**
Franken-Ender3

**Reproduction steps**
Enable output in the postprocessing script of changetz while having only a small low res LCD display on the printer :)

**Screenshot(s)**
- 

**Actual results**
The text is too long, and the useful information not shown

**Expected results**
useful information first, and/or custom text trunk

**Project file**
gcode snippet

`;:CAZD]
M117 Extruder 1 Temp: 240.0`

should be possible to named smth like
T1: 240

Ofc i can run a post-post script, but that little flexibilty or text reduction already  built-in would help more :)

best
JB

Ghostkeeper, 2020-12-23:
That current line is 22 characters wide. The Ender 3's display can show up to 24 characters in one line. Should be fine, right?
I'd expect that it'll wrap too.

The problem with `T1: 240` is that it's kind of code and people wouldn't understand that it's displaying the temperature. `T1` is g-code (which not everyone knows) and also refers to Extruder 2, not Extruder 1.

We could round it down to integers quite harmlessly I think. It'll only save 2 characters though.

muenznej, 2020-12-23:
Hey, thx for the reply.
Well it just truncates the text right before the temp and does not wrap around. At least not with my custom firmware(klipper).
I can totally see that it might be too specific, but maybe one could just make textfields for the optional LCD output  text within the changeatz script.
Besides this, is there any ways to copy existing entries in that script and past (repeat them?)

Ghostkeeper, 2020-12-28:
I see making the text on the LCD for various parts of the ChangeAtZ script as being too specific, indeed. Most displays should be able to show that text fine. There's also a different text for each parameter.

Too many configuration changes make an interface behave like a cockpit. Cockpits are generally fine for the pilots who venture into Github to request features, but make an interface very overwhelming to the average engineer. If hardly anyone will change a configuration option, I'd rather not provide it.

Post-processing scripts can be repeated, but there's not currently a way to copy-paste them in the GUI, no.

closed 2020-12-28 by Ghostkeeper

muenznej, 2020-12-29:
ok, i can understand your opinion.

Nonetheless, repeating entries in the GUI should be really useful to many more users than me, since retraction towers, temperature towers, speed towers and alike are probably used by quite some ppl.

should i make it an feature request or will you just say "no" ? :<

Ghostkeeper, 2021-01-06:
I'm not the only one that responds here. Or at least shouldn't be.

We've previously had a pull request that was [rejected](https://github.com/Ultimaker/Cura/pull/7816) though that implemented something similar. I found the PR worth considering so I discussed it with the development team but we rejected it after consideration. This pull request was a little different than your idea since it just appeared as an option to the ChangeAtZ script, which makes the ChangeAtZ script more complex. A "duplicate script" button in the post-processing window wouldn't be very intrusive or hard to understand, I think, so that could be a solution to the cockpit argument. The other issues with that PR would still apply for your idea though (limited value as a test due to the different circumstances at various heights above the build plate, and calibration being out of scope for Cura). I think my opinion would be that yes it should be added to Cura if the implementation is not intrusive, but it's not going to get priority. It's more of a "nice to have".

muenznej, 2021-01-06:
Sounds good. Ofc its not a priority thing but it would be really helpful. Thanks for considering it :+1: