Ultimaker/Cura #8969: cracks generated by cura 4.8.0

annemarietagant, 2020-12-18:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)
[image]

annemarietagant, 2020-12-18:
Application version
cura 4.8.0
Platform
HP pavillon gaming laptop 15,  windows 10 Famille version 1909 version du système d exploitation 18363.1256
Printer
Sovol SV02

Reproduction steps
just slice
Screenshot(s)
[image]

Actual results
(What happens after the above steps have been followed.)

Expected results
a nice surface!

Project file
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

Log file
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

Additional information

closed 2020-12-18 by annemarietagant

Ghostkeeper, 2020-12-23:
(Disable coasting.)