Ultimaker/Cura #8976: [4.8] Difficult choice of models

userosos, 2020-12-19:
Sometimes i cant select an block supports or an other elements. Can you add functions for it? When I hover my mouse over an item and press tab, the active item to move/scale, etc. starts to cycle through (becoming active sequentially).
[image]

nallath, 2020-12-21:
As a bit of a workaround, you can also use the model selector in the bottom left corner to select objects.

userosos, 2020-12-21:
> 
> 
> As a bit of a workaround, you can also use the model selector in the bottom left corner to select objects.

I am aware of this method. I have a very large number of blockers, so it is a very slow process. And I didn't find it there for some reason (most likely because of the awkward panel).

Vandrasc, 2020-12-22:
You can use the Object List feature, clicking on different object names from that list (pretty much all objects including mesh helpers on your buildplate will be listed there) will only highlight/select the wanted one. 

[image]

nallath, 2020-12-22:
> You can use the Object List feature, clicking on different object names from that list (pretty much all objects including mesh helpers on your buildplate will be listed there) will only highlight/select the wanted one.
> 
> [image]

That's the one I already suggested. But the way it currently works is that you can't select erasers in groups. It also gets a bit annoying if yo have like 50 of them (and they have weird names)

Ghostkeeper, 2020-12-23:
You'd like to be able to select objects that are behind other objects, I understand? Is that what you're asking for?