Ultimaker/Cura #8977: [4.8.0] Cura puts fails to put T1 code after printer start code and before layer 0 when using 2 extruders and E2 prints first

bellzw, 2020-12-20:
**Application version**
4.8.0

**Platform**
Windows 10 64-bit running on a Lenovo Yoga

**Printer**
BIBO2 Dual

**Reproduction steps**
1. Load attached 3mf project (Normal 0.15 mm, E1 printing PVA support, E2 printing PLA model)
2. slice the model
3. Examine the G-code

**Screenshot(s)**
[image] - half-way through the first slice showing that E2 is supposed to start the print
[image] - photo of how the print actually started  Note that it's the PVA starting the first layer.  The 2nd layer is printing with E2, as it should.

**Actual results**
G-code has T1 as the first executable code at line 12.  There happens to be a T0 in the printer starting code so the printer begins layer 0 using T0.  

**Expected results**
Layer 0 should start printing with E2 (T1).  See the code immediately following line 1759 that starts layer 1, and note that there is a T0 at line 1485 at the start of E1's rings in the skirt around the prime tower.

**Project file**
[Base test.zip](https://github.com/Ultimaker/Cura/files/5720484/Base.test.zip) - 3mf project file and the G-code it generated.

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
Cura puts some code (lines 12 - 21, mostly to set/get extruder and bed temperature) before the printer's start G-code.  Perhaps these lines that set temperature and T1 should all go after the start code?

nallath, 2020-12-21:
That is because the start g-code of the machine has T0 set in it. 

The machine definition has 
```
G21        ;metric values
G90        ;absolute positioning
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G1 Z2.0 F400 ;move the platform down 15mm
T0
G92 E0
G28
G1 Y0 F1200 E0
G92 E0
M117 BIBO Printing...
```
defined as it's start g-code.

bellzw, 2020-12-21:
> printer start code and before layer 0 when using 2 extruders and E2 prints
> first (#8977)

Well, yes, I know that there is a T0 in the start g-code; I mentioned that
in my description of the problem!  That's why in the example I provided, the
print starts with the wrong extruder unless I manually add a T1 after the
start g-code.  It is only an issue if Cura decides/needs to start with an
extruder different than what is last mentioned in the start g-code.

 

I do know that if I remove the T0 from the manufacturer's recommended start
g-code, the two extruders lose their alignment and prints will fail; E2 will
print about 3.25 cm to the left of E1; the two extruders are physically
separated by about that distance (see the "underside of extruders. png"
file).  The file 1203200930.png shows a print using E2 only and WITHOUT the
manufacturer's T0 in the start g-code; Loaded 3mf.png shows that  Cura
thinks the objects should be centered on the build plate.  

 

I'm not sufficiently knowledgeable about g-code to understand why T0 \ G92
E0 \ G28 \ G1 Y0 F1200 E0 \ G92 E0  in the manufacturer's start code
establishes the alignment between the two extruders.  If you can explain it,
that would be greatly appreciated and maybe I or the manufacturer can
generate better start code.

 

Barring generating new start code, the essence of my suggestion is to put
the manufacturer's start code at the beginning of the g-code file, before
anything generated by Cura.  That way, Cura's code is not superseded by the
start code.  So I the easiest solution is probable to have the g-code file
look like

.

;Generated with Cura_SteamEngine 4.8.0

; beginning of start code

M82 ;absolute extrusion mode

G21        ;metric values

G90        ;absolute positioning

M107       ;start with the fan off

G28 X0 Y0  ;move X/Y to min endstops

G28 Z0     ;move Z to min endstops

G1 Z2.0 F400 ;move the platform down 15mm

T0

G92 E0

G28

G1 Y0 F1200 E0

G92 E0

M117 BIBO Printing...

; end of start code

T1

M140 S60

M105

M190 S60

M104 T0 S175

M104 S200

M105

M109 T0 S175

M105

M109 S200

G92 E0  ; here's where the original file picks up again

G92 E0

.

This way the T1 overrides the start-code's T0 and the correct extruder
starts the print.

 

I suppose a variation would be to start heating the bed and extruders prior
to executing the start code to save a little time.  In this case the
beginning of the file could look like

.

;Generated with Cura_SteamEngine 4.8.0

M140 S60  ; start heating the bed

M104 T0 S175  ; start heating E1

M104 T1 S200  ; start heating E2

; printer start g-code

M82 ;absolute extrusion mode

G21        ;metric values

G90        ;absolute positioning

M107       ;start with the fan off

G28 X0 Y0  ;move X/Y to min endstops

G28 Z0     ;move Z to min endstops

G1 Z2.0 F400 ;move the platform down 15mm

T0

G92 E0

G28

G1 Y0 F1200 E0

G92 E0

M117 BIBO Printing... 

; end of start g-code

T1

M105

M190 S60

M105

M109 T0 S175

M105

M109 S200

G92 E0 ; from original Cura g-code

G92 E0

.

Of course, if other manufacturers have start g-code that manipulates
temperatures, then the leading M codes need to go after the end of the start
g-code.

 

Presently, this quirk forces me to edit every g-code file manually when Cura
starts with extruder 2.

<https://github.com/Ultimaker/Cura/issues/8977#issuecomment-748988825> , or
unsubscribe
5IPBANCNFSM4VCY7OZA> .
<https://github.com/notifications/beacon/AMFTMUJBBUEK6IHZZKVXP43SV5IPBA5CNFS
M4VCY7OZKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFSSKTGI
.gif> 

 

  _____  

 
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e
mail&utm_content=emailclient?utm_medium=email&utm_source=link&utm_campaign=s
ig-email&utm_content=emailclient> 

Scanned by McAfee
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e
mail&utm_content=emailclient?utm_medium=email&utm_source=link&utm_campaign=s
ig-email&utm_content=emailclient>  and confirmed virus-free.

Ghostkeeper, 2020-12-23:
Cura puts the extruder selection and heating before the start g-code because most printers use the start g-code to get the print started, e.g. prime the extruder or level the bed. For those things it's handy to have the correct extruder already selected and heated.
If we change that, all start g-codes need to be changed to include `T{initial_extruder_nr}` and the heating commands. It's possible, but likely to break Cura completely for a lot of printers, especially printers that people added themselves to Cura. We're not going to wager that bet any time soon.

If for your print start sequence you need to switch to `T0` temporarily for some reason, that's fine but you need to set it to the start extruder again when that's done.

So you should just add `T{initial_extruder_nr}` at the end of your start g-code. It's not currently preparing for your print since it leaves the printer in the wrong extruder after it's done.

> The file 1203200930.png shows a print using E2 only and WITHOUT the
manufacturer's T0 in the start g-code; Loaded 3mf.png shows that  Cura
thinks the objects should be centered on the build plate.

Github doesn't support e-mail attachments so your images do not show up for us. Please post them on Github's website rather than through an e-mail reply.

> I'm not sufficiently knowledgeable about g-code to understand why T0 \ G92
E0 \ G28 \ G1 Y0 F1200 E0 \ G92 E0  in the manufacturer's start code
establishes the alignment between the two extruders.  If you can explain it,
that would be greatly appreciated and maybe I or the manufacturer can
generate better start code.

This g-code basically says:

* `T0`: Switch to the first extruder.
* `G92 E0`: Pretend that your feeder is at position 0.
* `G28`: Home the extruder, making sure the coordinate system is aligned with the end stops on the printer.
* `G1 Y0 F1200 E0`: Move to one side of the build plate. If your endstops are already at the side of the build plate this will not do anything.
* `G92 E0`: Pretend again that your feeder is at position 0. This will definitely do nothing.

So it aligns the extruder to the homing position through G28, and claims that whatever position the filament is now will be considered to have the filament unretracted in the nozzle tip. Homing is important to make sure the X and Y axes are aligned well if the printer has been turned off or significantly jostled, or steps were lost previously. The `G92`s could be important with dual extrusion, depending on how your extruder trains are set-up.

Again for visibility: We can't move the initial extruder switch to the end of the start g-code without breaking everyone's printers except yours. Instead, add `T{initial_extruder_nr}` at the end of your start g-code to make sure your start g-code doesn't leave the printer in the wrong state.

closed 2020-12-23 by Ghostkeeper

bellzw, 2020-12-23:
Thank you.  It did occur to me that moving the heating codes might cause
problems for other printers; I didn't know how many other printers would
care.

 

Yes, I can add the Tn as needed (I have done that simply by editing the
gcode file) or just make sure the print always starts with T0 when I'm using
both extruders.  The latter might be preferable since Cura keeps changes in
the start code from session to session and writes those changes to a 3mf I
might make during a session.

 

What you write about the g-code is what I thought after looking at on-line
sources of information.  The surprise to me was that when I removed the T0
from the start code, the BIBO printer did not know how to align the two
nozzles and printed with extruder 2 about 3.25 cm to the left of extruder 1.
It's not clear to me how or why T0 interacts with G92 or G28, other than any
extruder reference will be to the selected tool.  Perhaps when the T0 is not
present, and the most recent Tn code is Cura's T1, the printer thinks
extruder 2 is the one at the origin and the controller always thinks it's
extruder 1.

 

This issue is closed as far as I'm concerned.

 

IIHHANCNFSM4VCY7OZA> unsubscribe.
<https://github.com/notifications/beacon/AMFTMUJGPO7GK4BPURYBPODSWIIHHA5CNFS
M4VCY7OZKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFS42SXY

 
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e

Ghostkeeper, 2020-12-28:
> It's not clear to me how or why T0 interacts with G92 or G28, other than any
extruder reference will be to the selected tool.

Cura's expectation is as follows: The `E` axis is tracked individually for each extruder. So if you have `G92 E0`, `T0` and `G92 E0` the second `G92` still has an effect. The X, Y and Z axes are not individual for each extruder though, so `G28` homes the print head for all extruders at once; homing again makes the print head move again but should result in approximately the same outcome. The X, Y and Z axes indicate the print head position, not the nozzle position. So that means that calling `G28` would move the print head to position 0,0 but the nozzles may end up in different positions. For instance, the Ultimaker 3 printer has one extruder with a nozzle offset of 0,0 and another extruder with an offset of 18,0. As a result, homing the print head makes the right extruder move to position 18,0.

braden0202, 2021-01-10:
> Cura's expectation is as follows: The `E` axis is tracked individually for each extruder. So if you have `G92 E0`, `T0` and `G92 E0` the second `G92` still has an effect. The X, Y and Z axes are not [...]

This is Braden from BIBO.   @Ghostkeeper , can you kindly help us revise the start gcode in json format file in next version cura ? Thank you in advance!
Change the start gcode can solve this problem.
In the bibo2_dual.def.json file, we can revise the start gcode as this: "machine_start_gcode": {
"default_value": "M104 T0 165\nM104 T1 165\nM109 T{initial_extruder_nr} S{material_print_temperature_layer_0, initial_extruder_nr}\nG21 ;metric values\nG90 ;absolute positioning\nM107 ;start with the fan off\nG28 X0 Y0 ;move X/Y to min endstops\nG28 Z0 ;move Z to min endstops\nG1 Z2.0 F400 ;move the platform down 2mm\nT0\nG92 E0\nG28\nG1 Y0 F1200 E0\nG92 E0\nT{initial_extruder_nr}\nM117 BIBO Printing..."
},

bellzw, 2021-01-10:
@braden0202, I’m far from an expert in G-code, but I don’t think the 

M109 T{initial_extruder_nr} S{material_print_temperature_layer_0, initial_extruder_nr}

is correct.  I think it should be

M109 T{initial_extruder_nr} S{material_print_temperature_layer_0}

because {...} is replaced by a single number.  I've assumed your intent is to wait for the initial extruder used by Cura, regardless of whether it is T0 or T1, to reach 165.

But, it’s not obvious why you are setting hot end temperatures in the startup code since Cura sets them before the startup code is inserted; your M104 and M109 codes upset this. The startup code in the BIBO2 dual.def.json that I have from BIBO does not contain M104 commands before the G21 command; I don’t think they or the M109 make sense here.  

The problem is the T0 near the end of the startup code can change the active extruder.  I think you only need to put T{initial_extruder_nr} before the M117 to enable the extruder Cura first specifies in its G-code.

A potential issue, however, is that I don't know if Cura might have multiple Tx commands prior to the startup code, so there it might be possible for {initial_extruder_nr} refer to the wrong extruder. @Ghostkeeper will have to tell us if that can happen.  What might be a better solution, although I have no idea if it is feasible,  would be to have a way to save the current extruder number so that it could be saved at the start of the BIBO startup code and recalled for a T code just before the M117.  A next-to-last extruder number is not sufficient if the BIBO startup code has multiple T codes in it.  The idea is to leave the printer in the same T state after the startup code as it was at the beginning.

An alternative is to write a post processor, something I've though about doing, to look at the gcode file, find the last T code before BIBO start code, and insert the same code immediately after the M117.  That way it wouldn't matter how many different T codes were generated by Cura or BIBO's startup code.

bellzw, 2021-01-12:
I may have learned something today: I was perusing the fdmprinter json file
and ran across material_print_temp_prepend and material_bed_temp_prepend.
Since these are set to true, does this mean that if @braden0202 includes
M104 and M109 codes in the startup gcode, then Cura will not put the those
codes ahead of the start gcode?  If so, then my concerns are removed.

 

However, unless it is certain that the most recent T code before the start
gcode is initial_extruder_nr, then it is possible @braden0202's startup to
fail.  He really needs a token holding the last tool referenced by a T code
prior to the start of the printer's startup gcode.

 

FZNNANCNFSM4VCY7OZA> .
<https://github.com/notifications/beacon/AMFTMUL3S6B3BGNQMPP27FLSZFZNNA5CNFS
M4VCY7OZKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFUS3STY

 

 
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e

<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e

Ghostkeeper, 2021-01-18:
> I may have learned something today: I was perusing the fdmprinter json file
and ran across material_print_temp_prepend and material_bed_temp_prepend.
Since these are set to true, does this mean that if @braden0202 includes
M104 and M109 codes in the startup gcode, then Cura will not put the those
codes ahead of the start gcode?  If so, then my concerns are removed.

Almost. Those settings are automatically changed by Cura depending on whether any temperature-related setting is used in the start g-code (for respectively the nozzle temperature or the build plate temperature). So that means that if the start g-code contains `{material_print_temperature_layer_0}`, Cura will not output `M104` or `M109` before the start g-code.

Cura will always put that one `T` command before the start g-code. But it's always `T{initial_extruder_nr}` and always just the one.

So I think then Braden0202's start g-code is correct and reliable. It switches the extruder back and forth a bit (and multiple subsequent `T0` when `T0` is the starting extruder) but there's no harm in that. I'll trust that they've tested it on their printer.

@braden0202 I've applied your correction here: https://github.com/Ultimaker/Cura/commit/0a7dc34a9c5dde1f71f1fb3977b444ea4de745e4
It will be released with the next version of Cura.

tlhintoq, 2021-08-01:
Different symptom but I bet its related.

T1 is disabled.

Yet the GCode at line 22 Set T1 as the starting extruder **AND** used all the settings from T1 instead of from T0.  So wrong temp, line width, speed and so on.

Why would Cura choose to default the disabled extruder?
[image]

Ghostkeeper, 2021-08-09:
@tlhintoq It's not really related. Well, maybe sideways related because it's in the start g-code.

Both Cura front-end and CuraEngine back-end are seeing that the initial extruder is T1, as evident by `T1` being put there by CuraEngine and the settings being evaluated for the initial extruder by the front-end. The initial extruder is [derived from the settings](https://github.com/Ultimaker/Cura/blob/176077c910b9dabe76f5c2047e9b87759f4cef0d/cura/Settings/ExtruderManager.py#L277-L296). It'll use the adhesion extruder, the support extruder or the extruder of one of the models.

The adhesion extruder and support extruder should get set to (one of) the remaining extruder(s) when they are set to be printed with the extruder you disable. However it is a known problem that if you have a profile that overrides the extruder selection setting, reverting that setting change is still possible, and also discarding changes when going to that profile still causes the disabled extruder to be selected. Perhaps this is what happened for you.

We could only really tell if you provide reproduction steps, as in, create a proper issue report for us.

Ghostkeeper, 2021-08-09:
Oh, apparently it was posted as a separate issue too: https://github.com/Ultimaker/Cura/issues/10217