Ultimaker/Cura #8980: Cura 4.8 prints more walls than needed

EWMorrison, 2020-12-20:
**Application version**
4.8

**Platform**
Windows 10 64bit

**Printer**
Ender 5

**Reproduction steps**
1. Create a hollow model with walls thinner than the cura wall thinkenss setting
2. Slice

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Cura treats the wall thickness as a minimum, should be treated as a maximum. 

**Expected results**
Cura should maintain the modeled geometry. If the imported STL has a 3mm wall thickness in the part, Cura should not add material. The user should change the STL if they desire different geometry. If the wall thickness of the part exceeds the Cura wall thickness setting, the part should have the wall thickness chosen in Cura and infill, if selected.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
This likely only applies to hollow parts. In my case these were tubular parts that were being printed with a reduced inner diameter because Cura was maintaining the wall thickness setting instead of recognizing that this was changing the part geometry.

nallath, 2020-12-21:
I don't understand what you mean. Cura treats the wall thickness as maximum. 

I've sliced the following object:
[image]

As you can see, in cases where the wall_thickness is more than the thickness than the model, fewer walls are used:
[image]

If you have a specific model where this is going wrong, please attach it (or provide some pictures that clarify what you mean)

EWMorrison, 2020-12-21:
Jamie,
The attached STL produces a cylinder instead of the tube as it is drawn. If
I allow infill it will make it solid. It will make the wall thickness equal
to what I set in cura. Prusa slices this correctly. This was drawn with
Fusion360. The part was designed to be 3 lines thick. I have a part where
the 3 lines becomes 6 then back to 3. Cura will make it 6 the entire height
because that is what wall thickness is set to.

[image: image.png]
[image: image.png]

[image: image.png]

Regards,
Eric

nallath, 2020-12-21:
Sending images as part of an attachment don't work for github. You will have to upload them via the website (or host them somewhere)

Could you also share the model?

EWMorrison, 2020-12-21:
[image]
[image]
[image]

EWMorrison, 2020-12-21:
[TestRing.zip](https://github.com/Ultimaker/Cura/files/5725983/TestRing.zip)

nallath, 2020-12-22:
I can't reproduce your issue. Are you sure that you didn't enable a setting like "horizontal expansion"?

See my attempt:
[image]

If you can share a project with me, I can see all the settings that you use to print this.

EWMorrison, 2020-12-22:
So I went through the profile line-by-line and turning off "Remove All Holes" has fixed the problem. I don't remember turning this on or understand why I would, but that seems to be the answer. Thank you for working through this with me.

closed 2020-12-22 by EWMorrison

Ghostkeeper, 2020-12-23:
Your confusion arises from the fact that Cura will create 6 walls for you from all sides of the model. That is, both the inside of your ring and the outside of your ring. So that is 12 walls in total, and if the ring is thick enough it would create 6 walls, then infill, then 6 walls again. After all, when printing a normal ring, people don't want their infill to be visible on the inside of their ring.

Remove All Holes removes the inside opening in your model. So that eliminates half of the walls and closes the opening entirely.