Ultimaker/Cura #8981: Variable print speeds based on layer number and XY coordinates

mshaefer, 2020-12-20:
**Is your feature request related to a problem? Please describe.**
Not necessarily, except that I have a print that needs letters/numbers (a prototype) and in order for the letters/numbers to be legible I have to print at a speed and layer height that makes the whole project take longer.

**Describe the solution you'd like**
To be able to select (perhaps in something like the Preview view) a given range of layers (say "from layers 12 to 16") and all points within certain XY parameters (basically like within a selection window), and tell the print to go faster or slower in that particular selection.

**Describe alternatives you've considered**
I've tried editing the Gcode myself by increasing the print speed and feedrate between layers, but the major benefit really comes from being able to alter the layer height within certain parts of a print.  When doing this layer by layer it becomes incredibly tedious.  I've resorted to basically setting up CSV excel files that will automatically calculate the z value so the machine will "speed up" and "slow down" at the correct z value for the given layers.  It's incredibly tedious, but saves me time in the long run because I've got to produce dozens of the same prototype.  In my example it's kind of simplified because the first 23 layers are mostly just flat, then the next 9 include the writing and thus require finer detail than what I need everywhere else, and then the rest of the 190 layers can go on as fast as possible.

I can imagine countless scenarios where it would be enormously helpful to be able to just say "go as fast as possible up unilt layer 9 and then for this part of the print" or "for all features of the print within this particular area", "print nice and slowly and only do that from layers 10-20", and then "after that, go back to printing as fast as possible".  It would make it so I can have speed without detail where I need it, and detail without speed where I need it, all within the same print and just with some simple (albeit tedious when done by hand) Gcode editing.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

Thanks for taking this request.

nallath, 2020-12-21:
All bits, except for the "for all features within this particular area" are possible with the ChangeAtZ post-processing script.

Alternatively, you could also have a look at modifier meshes. They allow for different settings to be applied to different parts of the object. The downside of this approach is that walls will be "cut up" by this approach.

Ghostkeeper, 2020-12-23:
If the modifier mesh is applied to the entire layer, there won't be any walls to cut up anyway ;)

Ghostkeeper, 2020-12-23:
I'd consider this already implemented through the ChangeAtZ script then.

closed 2020-12-23 by Ghostkeeper

mshaefer, 2020-12-24:
I got it, but was wondering whether there was a way to change things like
layer height as well.  So far it seems like change at z will really only
let me slow down / speed up a print for certain layers / z heights.  Seems
like it should be possible to modify the g code so that these adjustments
were available to adjust at all layers.  In other words, say print layers
1-13 at 0.4mm, 140 mm/s, 205c; and then change at z (or layer) and print
layers 14-20 at 0.2mm, 60 mm/s, 210c; and then layers 21-120 at 0.4, 100
mm/s, 205.  Get it?  In that way we'd be able to tune prints with a LOT
more freedom and save a ton of time on the really basic parts while not
sacrificing quality on the more detailed parts.

-- 

*Mark M. Shaefer*

*Office of the District Attorney, Augusta Judicial Circuit*

*735 James Brown Blvd, Suite 2400*

*Augusta, GA 30901*

*P:  706-821-1135*

*F:  706-821-1237*

*This e-mail contains confidential information and is intended only for the
individual named. If you are not the named addressee, you should not
disseminate, distribute or copy this e-mail. Please notify the sender
immediately by e-mail if you have received this e-mail by mistake and
delete this e-mail from your system. The City of Augusta accepts no
liability for the content of this e-mail or for the consequences of any
actions taken on the basis of the information provided, unless that
information is subsequently confirmed in writing. Any views or opinions
presented in this e-mail are solely those of the author and do not
necessarily represent those of the City of Augusta. E-mail transmissions
cannot be guaranteed to be secure or error-free as information could be
intercepted, corrupted, lost, destroyed, arrive late or incomplete, or
contain viruses. The sender therefore does not accept liability for any
errors or omissions in the content of this message which arise as a result
of the e-mail transmission. If verification is required, please request a
hard copy version. AED:104.1*

Ghostkeeper, 2020-12-28:
Aside from Adaptive Layers, this is indeed not possible.
Practically, the post-processing script gets g-code as input, so it doesn't have access to the 3D model and can't re-slice parts of the scene.
Outside of post-processing scripts your feature request is then duplicate of https://github.com/Ultimaker/Cura/issues/4235. Per-layer settings are practically very difficult and also intuitively hard to understand for novices.

mshaefer, 2020-12-28:
What I'm saying though is that I've successfully done this by hand by
modifying the gcode z height layer by layer, changing z heights and layer
thicknesses, but it's painstakingly slow and it works.  What I can't tell
from the prints I've done is whether the "squish" (i.e. the thickness of an
individual line) is affected by the layer height.  I just can't tell.

This is how I'm accomplishing this.  Let's say I've got a print that is
12mm tall. I want to print the first 20 layers at 0.2mm layer height and
then the rest of the print at 0.4mm layer height.  I slice the gcode at a
0.2mm layer height and 0.4mm layer height.  I then go through each layer
AFTER the 0.2mm layers one by one and change the extrusion values and the
layer heights for each layer.  In other words, I slice it 0.4mm so that i
can get good extrusion values, and then the z height is just simple math.

So 0.2mm for 20 layers taking me to 4mm, and then for the next 30 layers I
print at a .4mm layer height with a very slightly higher extrusion value.
I guess I am still confused why this isn't possible in the program, setting
layer heights (within reason), between different z heights to accomplish
different resolutions and thus different speeds.

-- 

*Mark M. Shaefer*

*Office of the District Attorney, Augusta Judicial Circuit*

*735 James Brown Blvd, Suite 2400*

*Augusta, GA 30901*

*P:  706-821-1135*

*F:  706-821-1237*

*This e-mail contains confidential information and is intended only for the
individual named. If you are not the named addressee, you should not
disseminate, distribute or copy this e-mail. Please notify the sender
immediately by e-mail if you have received this e-mail by mistake and
delete this e-mail from your system. The City of Augusta accepts no
liability for the content of this e-mail or for the consequences of any
actions taken on the basis of the information provided, unless that
information is subsequently confirmed in writing. Any views or opinions
presented in this e-mail are solely those of the author and do not
necessarily represent those of the City of Augusta. E-mail transmissions
cannot be guaranteed to be secure or error-free as information could be
intercepted, corrupted, lost, destroyed, arrive late or incomplete, or
contain viruses. The sender therefore does not accept liability for any
errors or omissions in the content of this message which arise as a result
of the e-mail transmission. If verification is required, please request a
hard copy version. AED:104.1*

Ghostkeeper, 2020-12-30:
Oh it's definitely possible in theory. Adaptive Layers does it, albeit automatically rather than allowing the user to change thicknesses manually.
The problem is that there is no interface to adjust settings per layer.

mshaefer, 2020-12-30:
Precisely. That’s what I was asking about as a feature request, some sort of interface in cues to adjust layer thickness per layer. I appreciate y’all fielding all my questions.

Ghostkeeper, 2021-01-07:
I think that discussion is best taken into #4235 then. It's not something that I'd expect we'll develop soon. But it's good to get all the feedback in when we're ready. Or when someone else in the open source community wants to give a shot at it in their Cura editions or perhaps even to eventually submit it to us (although that is quite a big change to accept from outside).

SHREYAS1188, 2025-04-07:
Hi @mshaefer I know the thread is very old its been 5 years, but i want to do the same thing which you wanted to do. If you are still up can we pls talk on this. Looking forward to hear on this @mshaefer