Ultimaker/Cura #8983: unsuported parts on the inside of a print

TheAxip, 2020-12-20:
**Is your feature request related to a problem? Please describe.**
Hi,

I was scrubbing thru my model which I'm printing for the last 8 hours and found this pice which will be printed in thin air and is unsupported by infill. I know the solutions to prevent this, I just didn't realise at the time that there was a problem.

**Describe the solution you'd like**

A warning like "check for unsupported part at layerhight xx" would draw the attention to such locations.

**Describe alternatives you've considered**

You could fix these parts directly with additional support structures but I think drawing attention would be enough to resolve the oversight and enable the user to resort to his/her prefered solution.

thanks for listening
**Additional context**
[image]

konskarm, 2020-12-21:
In theory that sounds like it could work, but in practice this can be very annoying. If you have multiple places where something isn't supported you may end up having literally hundreds of messages on your screen. In addition, the model can have very tiny unsupported parts (e.g. a tiny unsupported tip), so displaying something like "Check layer x" will end up being confusing if it is not immediately visible.

Ghostkeeper, 2020-12-24:
This is really a symptom of using too low infill density, and there will always be combinations of settings that ruin your print.

It would be good if Cura were able to automatically handle cases like this. And in some ways it can, with Gradual Infill for instance, or Skin Edge Support Thickness. In the future we're also looking into using a tree support like structure to produce infill.

jhyslop, 2021-02-04:
@Ghostkeeper it doesn't necessarily happen with infill. I have the same problem. In my case, the infill is set to 100% - can't get much higher density than that :)

I have found two unsupported areas (there may be others, I haven't searched). One is part of the support tree itself, and the other is part of the model itself. The attached .zip file contains the .stl source file, the cura settings, the gcode output, and a few screen shots to show where the problem is.
[cura tree support defect.zip](https://github.com/Ultimaker/Cura/files/5923487/cura.tree.support.defect.zip) I'm running version 4.8.0

I think this is a bug - the bottom of model is completely unsupported, and there is no reason it can't be supported. Should I open a separate ticket for that, since this ticket is for a feature request?

Julius971, 2021-02-10:
Hey, found this issue while being bored. Wouldn't something like "infill support" maje the deal? The slicing process would look for those parts before everything else and add geometry as support. (Maybe with support settings applied later on).
After that the usual slicing process generates the infill around the inner support structure.
This should be the easiest implementation with best results.

Another idea could be cura adding cobweb like structures on a few infill layers under the floating part. This would try to keep most of the desired infill structure. This could be turned on and off by an option like infill maximum point distance. 
This should add the web under a part, if the infill won't provide enough surface.

Ghostkeeper, 2021-02-11:
> @Ghostkeeper it doesn't necessarily happen with infill. I have the same problem. In my case, the infill is set to 100% - can't get much higher density than that :)

Your support infill density is set to 0% though, so it kind of makes sense that you could get unsupported parts. However your case is quite separate from this one. It's really more like issue #4342.

> Hey, found this issue while being bored. Wouldn't something like "infill support" maje the deal? The slicing process would look for those parts before everything else and add geometry as support. (Maybe with support settings applied later on).
After that the usual slicing process generates the infill around the inner support structure.
This should be the easiest implementation with best results.

Yeah I would consider the Infill Support feature to be a workaround for this issue, if you combine it with an Extra Infill Wall. In case you weren't aware, this [feature already exists](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill/infill_support_enabled.md) in part. Though you are also speaking of something that sounds more like [Skin Edge Support Thickness](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill/skin_edge_support_thickness.md).

> Another idea could be cura adding cobweb like structures on a few infill layers under the floating part. This would try to keep most of the desired infill structure. This could be turned on and off by an option like infill maximum point distance.
This should add the web under a part, if the infill won't provide enough surface.

Effectively this works because it increases the density of the infill. In your idea it increases it irregularly, but increasing the density regularly provides better guarantees. Effectively you're talking about Gradual Infill then.

Julius971, 2021-02-19:
Gradual infill would't solve all the problems. And it would't it cause more iregularities to the infill as a single needed structure/web/whatever to support inner layers. And even with gradual infill there's no guarantee that everything is properly supported.

Ghostkeeper, 2021-03-01:
Sorry, I was assuming that by a "cobweb" you meant random lines through the layer. The random lines would definitely (almost by definition) be more irregular than a pattern and would provide fewer guarantees as to the distance between supported points.

If you didn't mean random lines, then it must be a pattern of some kind. Maybe you meant [radial infill](https://github.com/Ultimaker/Cura/issues/8811) then?

Of course there is no guarantee that everything is properly supported unless you support everything, i.e. 100% density. Instead, Cura can support the edges (Skin Edge Support feature) and then provide guarantees on the distance between supported points. Lines infill is as good as it gets to minimise the distance between supported points, if the lines are perpendicular to the direction of the skin lines.

Zackorion15, 2021-03-15:
> Hey, found this issue while being bored. Wouldn't something like "infill support" maje the deal? The slicing process would look for those parts before everything else and add geometry as support. [...]

Honestly this is exactly the option I would love to see.

A two stage infill generation option, where the first stage generates "supports" on the interior of the model where needed to ensure all above features print well, then the second pass builds general infill around that for general part strength.

>Yeah I would consider the Infill Support feature to be a workaround for this issue, if you combine it with an Extra Infill Wall.

The issue with the existing infill support option, is that while it does great generating all the needed internal support areas, it totally turns off infill in all other areas.

christoftaylor, 2021-09-22:
I'm having similar issues with a print and came up with three suggestions while I was watching it print in mid-air in between the infill...

Supports -- the routine that builds supports only looks outside the model. If that routine also looked inside the model, it could add supports in between the infill for when upper layers overhang the gaps. This seems like the cleanest solution, and seems like it would require the least amount of thought since it would just be refactoring the support logic instead of building new logic. 

Skin aware of where the infill is -- in the model I'm printing, the upper layers are getting a layer of skin that it is then building upon. The size and shape of that skin is based on what it is about to build above it rather than what supports are below it. If it laid the first layer of this skin from infill wall to infill wall, it would then be supported without printing into the voids. 

Infill density -- if the slicer knows that it needs to print mid-air on an upper layer, it could automatically increase the infill density a few layers below, to create a supported floor that it could build up from. This is similar to the above, but applied in the infill logic instead of in the skin logic. 

There are probably other ways to solve this if I booked a room at Holiday Inn Express. I definitely believe it's a problem that the slicer should solve without needing me. If it can't solve it, if it detects that it is air printing, it should raise a warning so the user knows their is a problem with the print before they are half way in. 

I appreciate you looking at it and being so open to the all the suggestions above and in other threads. You guys rock!

Ghostkeeper, 2021-09-29:
> Supports -- the routine that builds supports only looks outside the model. If that routine also looked inside the model, it could add supports in between the infill for when upper layers overhang [...]

This exists. It's regulated by a setting called Infill Support. It's not a complete solution, because yes, it will cause the infill to be supporting everything that needs support, but it still supports it with a certain sparse infill pattern. It reduces the total amount of infill though, so you could increase the infill density to arrive at a similar printing time. It's mentioned earlier in this thread as a workaround.

> Infill density -- if the slicer knows that it needs to print mid-air on an upper layer, it could automatically increase the infill density a few layers below, to create a supported floor that it [...]

This exists too, known as Gradual Infill.

> Skin aware of where the infill is -- in the model I'm printing, the upper layers are getting a layer of skin that it is then building upon. The size and shape of that skin is based on what it is [...]

I think this is what the original author intended with this issue, really. In theory it's possible although not without extra flow interruptions. In practice, it's likely very complex to build because you would have to make something that works for each of the dozen infill patterns. Where to find and how to connect to the nearest infill line will depend heavily on the infill pattern and how it's rotated.

Another possible solution is the [Lightning Infill](https://github.com/Ultimaker/CuraEngine/pull/1395) pattern. You could increase the infill density to 100% then and it'll still be mostly empty below.

Ghostkeeper, 2022-04-10:
What do you think of gradual infill as a solution then? It holds a baseline infill density at the step limit (e.g. 10%) but creates a higher infill density at the top to support the skin better.

GregValiant, 2025-07-03:
There have been improvements in this in the latest versions.  Prior to the infill completing, extra lines get laid to support the following skins.

Can this be closed?