Ultimaker/Cura #8984: [4.8.0]Intermittent freeze when trying to save gcode file

bellzw, 2020-12-20:
**Application version**
4.8.0

**Platform**
Windows 10 64-bit running on Lenovo Yoga 14

**Printer**
BIBO2 Dual

**Reproduction steps**
1. Slice a model (doesn't seem to depend on the model, but it has failed with the attached stl and 3mf)
2. Save gcode file

**Screenshot(s)**
[image]

**Actual results**
Cura stops responding and I must close it, restart it and hope I can save the gcode.  Cura seems to remember the profile and settings from just before the freeze, including settings I've changed with respect to the profile.

**Expected results**
File dialog box lets me select a file name and save the gcode file.

**Project file**
Unfortunately, I didn't get far enough to save a 3mf before the freeze.  This 3mf was saved after restarting Cura and saving it before trying to save the gcode.  Save to file was successful.  Maybe after the 3mf was saved, future gcode saves work?

I've included the stl, project file, and gcode that were eventually saved by Cura in the zip.
[Base test.zip](https://github.com/Ultimaker/Cura/files/5721666/Base.test.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5721650/cura.log)

**Additional information**
This is an intermittent problem and usually a restart of Cura resolves the it, but sometimes it requires a few restarts.  In this case, Cura is trying to access a 3 TB USB disk.  When Cura hangs up, I am still able to access this disk via Windows, so I don't think it is caused by the destination being a USB disk.

rburema, 2020-12-23:
Does saving large files to t he disk from other programs work?

bellzw, 2020-12-23:
Yes.  I can save large files with other programs.  A file of several or tens
of Mbytes is not that large for a 3 TB disk.  The disk is "smart" in the
sense that when has not been accessed for some minutes, it spins down, so
there can be a delay while it spins up again.  It's an intermittent problem
in 4.8.0, and often, but not always, closing Cura and restarting it lets me
save the gcode file.  I have not done a careful study of whether it only
happens when I save gcode files.

gcode file (#8984)

 

<https://github.com/Ultimaker/Cura/issues/8984#issuecomment-749991890> , or
unsubscribe
GOGFANCNFSM4VDPCQ3Q> .
<https://github.com/notifications/beacon/AMFTMUIPTSUDALEH4F66FJTSWGOGFA5CNFS
M4VDPCQ32YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFSZ7PUQ
.gif> 

 

  _____  

 
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e
mail&utm_content=emailclient?utm_medium=email&utm_source=link&utm_campaign=s
ig-email&utm_content=emailclient> 

Scanned by McAfee
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e
mail&utm_content=emailclient?utm_medium=email&utm_source=link&utm_campaign=s
ig-email&utm_content=emailclient>  and confirmed virus-free.

Ghostkeeper, 2020-12-24:
That file dialogue is created by your operating system, through Qt, and we have relatively little control over it from our application. I can imagine that a disk spindown can cause it to freeze for a few (<5) seconds as the disk spins up again. That used to happen to me too until I made it spin down less aggressively. But then it would happen for all applications that use the disk. For me it happened also with the Ubuntu file explorer, Firefox accessing its cache, everything. And it shouldn't be too long.

Maybe we can see if other people have the same problem and discover what the common denominator is between the users where it freezes for longer than a few seconds.

bellzw, 2020-12-24:
I notice that other applications will freeze for the few seconds it takes to
spin up the drive, but it's only for a few seconds as you mention.  When
Cura freezes, I give up after about a minute, so I really don't know if it
will recover at some point.  Next time it happens, I will set a timer, and
post what I find.

 

I have Python 3.9 and wxPython 4.1.1 installed on my computer, although I
was not using either when Cura froze..  Maybe that's a clue?

gcode file (#8984)

 

MXIZANCNFSM4VDPCQ3Q> .
<https://github.com/notifications/beacon/AMFTMUOYWDNVYQWSODZ27PDSWMXIZA5CNFS
M4VDPCQ32YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFTAVO3I

 

 
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e

<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e

bellzw, 2020-12-27:
[cura.zip](https://github.com/Ultimaker/Cura/files/5743966/cura.zip)
This zip contains cura.log, cura.cfg, and my own log of dates, times, and observations for the latest freeze.   The laptop continued to operate, and still does.  After the freeze, I left Cura running for 21 hours, but it never recovered.  Hope this helps.

Ghostkeeper, 2020-12-28:
Cura doesn't use Python installations on your system. It comes built-in with its own Python installation. And it doesn't use wxWidgets at all (we use Qt).

Ghostkeeper, 2020-12-28:
The log file doesn't seem to contain any information relevant to this issue as far as I could find. That sort of makes sense. Cura instructs Windows to open that file dialogue, and then the Windows File Dialogue freezes the application for some reason.

Nobody else has seen the issue so far as far as we know and it doesn't occur in our own testing either. This makes it hard to fix; we can't get debug information from Qt, and we can't test if things we change fix the issue.

bellzw, 2020-12-28:
If I'm the only one who's reported this, then it's probably reasonable to
assume that it's my laptop and possibly nothing related to Cura.  I guess
I'll just need to keep track of this to see if I can correlate it to
anything else the computer is doing.  So far, all other running programs
continue to run, and the USB disk remains accessible.

 

I mentioned Python and wxPython in the hope that that was a distinguishing
feature of my system.  If that's not it, then that's not it.

 

You can close the issue or not, as you see fit.

gcode file (#8984)

 

C6PJANCNFSM4VDPCQ3Q> .
<https://github.com/notifications/beacon/AMFTMUIOXT2W54RZ5JLUFW3SXC6PJA5CNFS
M4VDPCQ32YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFTHY6BQ

 

 
<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e

<https://home.mcafee.com/utm_medium=email&utm_source=link&utm_campaign=sig-e

davidgilson, 2023-07-06:
I'm having the same issue in Cura 5.3.1

davidgilson, 2023-07-06:
Returning to say that I'm having the same issue in Prusa Slicer 2.6.0 too.

bellzw, 2023-07-08:
@davidgilson  Most likely it's an issue with the computer/operating system.  My laptop (Lenovo Yoga 14, ~5 yrs old) does this sometimes when downloading or copying files, and other times when doing things unrelated to Cura, like using Acrobat Reader, or Microsoft Office products.

GregValiant, 2024-11-20:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

bellzw, 2024-11-21:
Yes, this can be closed w.r.t. 4.8 a) because I’ve moved past that version a while ago, and b)  I’m having no problems with a newly installed 5.9.0.

closed 2024-11-21 by GregValiant