Ultimaker/Cura #8985: [4.8] "Initial Layer Horizontal Expansion" has no effect in negative direction

877dev, 2020-12-21:
**Application version**
4.8

**Platform**
Windows 10 Pro x64, 8700K CPU, GTX1080 GPU

**Printer**
Creality Ender-3 Pro

**Reproduction steps**
Using standard 4.8 profile, adjust "Initial Layer Horizontal Expansion" in order to reduce elephants foot

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
Set to 0
[image]

Set to +1
[image]

Set to -1
[image]

**Actual results**
The actual print comes out with elephants foot, always the same no matter what value is used.

**Expected results**
Reduction of elephants foot when choosing -1

**Project file**
https://www.dropbox.com/s/1uskms1ezs4el1k/CE3PRO_Cura%20Flow%20Rate%20Extruder%20Calibration%20cube%204mm_nozzle.3mf?dl=0

**Log file**
https://www.dropbox.com/s/3zfygy98owg4do9/cura.log?dl=0

**Additional information**
n/a

nallath, 2020-12-21:
This isn't a Cura bug. The initial horizontal expansion is applied correctly, as you've already shown in your screenshots.

closed 2020-12-21 by nallath

Ghostkeeper, 2020-12-24:
Elephant's foot is not just a matter of the initial layer. In your case it looks like it's caused by the material being able to shrink where it's just thin walls, but not shrink in the flat base of 100% density skin of the model.