Ultimaker/Cura #8988: Cura 4.7 and 4.8 crash on startup

Pritolus, 2020-12-21:
**Application version**
4.7 and 4.8.0, same issue

**Platform**
Windows 10 Build 18363
GTX 1070 Ti (Asus GTX 1070 Ti Strix OC)

**Printer**
N/A

**Reproduction steps**
1. Launch Cura
2. Watch it finish starting up and then crash

**Screenshot(s)**
N/A

**Actual results**
Cura crashes. Every time.

**Expected results**
Cura should launch.

**Project file**
N/A

**Log file**
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/5724723/DxDiag.txt)
[cura.log](https://github.com/Ultimaker/Cura/files/5724724/cura.log)

**Additional information**
Worked fine for months, didn't use it for a while, then it wouldn't start.
When you launch Cura you get the small gray progress box, telling you all it's doing while preparing to launch. Then it disappears, and the main Cura window opens, but in all white with no info in it. It will barely have time to pop up before it disappears, along with any Cura related process in task manager. Happens 100% of the time, even after updating Cura from 4.7 to 4.8.0 and after running Windows Update as far as it would go. Deleting the config folder (C:/Users/user/AppData/Roaming/cura) made no difference. So far nothing I've done has gotten it to actually launch.

Vingen168, 2020-12-22:
i have the exactly the same issue,  but after reinstalling cura 4.6 the 4.8 works again.

me-ga-pixel, 2020-12-22:
Same issue. Reinstalled 4.7 and 4.8 without success. I have tried installing Cura 4.6.2, it starts correctly, but 4.8 continues crashing.

fabriciofx, 2020-12-23:
The same problem here!

rburema, 2020-12-23:
This is a duplicate of:
https://github.com/Ultimaker/Cura/issues/8892
Please continue any discussion there (closing this in favour of that).

We see this happening with multiple monitor setups a lot lately:

There is a workaround (originally published here, different ticket though, please use the above one to continue discussion!):
https://github.com/Ultimaker/Cura/issues/8904
that states the following:

> Known Fix:
> Set 'window_left' to any positive value in [general] section of %appdata%\cura\4.8\cura.cfg

closed 2020-12-23 by rburema

fieldOfView, 2020-12-23:
A better fix is to remove the 'window_left' line in [general] section of %appdata%\cura\4.8\cura.cfg, and then on the next (now hopefully succesful) launch of Cura be sure to go the the General preferences and uncheck the "Restore window position on start" checkbox.

rburema, 2021-01-06:
@fieldOfView Sorry, I keep forgetting that.