Ultimaker/Cura #8990: [Feature Request] Minimum Infill Layers setting

KenLucke, 2020-12-21:
**Is your feature request related to a problem? Please describe.**
 I have had several models of late that end up generating one layer (sometimes even a thin 0.1mm layer) of infill. This causes the next layer to be a slow "bottom" layer. It actually wastes more time (it seems) to have that happen than to just print a normal layer, and he amount of filament saved, in most cases, is minuscule.

**Describe the solution you'd like**
 A setting under infill that would require a minimum thickness or number of infill layers (defaulting to 2*layer height or something) before the program would actually switch to considering the layer to be infill.

**Describe alternatives you've considered**
 Support blocker in the layer set to 100% infill, which is a PITA to do and get positioned correctly.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

Ghostkeeper, 2020-12-24:
> than to just print a normal layer

With a "normal layer" you mean that it prints skin?
But wouldn't that cause another slow "bottom" layer to be printed then instead of the faster infill?
I'm not sure if I understand what you intend the infill to be replaced by then.

KenLucke, 2020-12-24:
Say it's printing along somewhere in the middle of a 20 mm thick model, doing its normal body stuff.  It then hits a section where it thinks it needs infill, for one layer, then prints a "bottom" or "skin" layer, then continues on its merry way as it did before (I haven't done this, but theoretically make a model 20.2mm thick, with top and bottom thicknesses of 10mm and a layer height of 0.1mm or 0.2mm - between the top and bottom thicknesses, [theoretically] it would try to do one or two layers of infill, which is silly.  If you set a "Minimum Infill Height" setting, as the feature requests, it would just ignore that tiny gap and keep printing as it had the previous layers).

It should just keep going with previous layers if the thickness of the area that was going to be infilled is less than the Minimum Infill Thickness.

Ghostkeeper, 2020-12-28:
Right. So instead of infill, it should be printing whatever is above or below the infill.
But of course, the only thing that can be above or below infill is skin, so the feature request is to print skin instead of infill if the infill is thin enough (vertically). Thanks for the clarification.

Ghostkeeper, 2020-12-28:
I don't really see the harm in printing a single layer of infill though. Having a single layer of infill is somewhat weaker than making it 100% solid skin all the way through, I can see that, but the Top/Bottom Thickness setting should be such that the skin has sufficient thickness for the application's strength requirements.

Your stated problem is that it wastes time printing a single layer of infill, but the proposed solution wastes more time than printing the infill. I'm very confused! :confounded:

KenLucke, 2020-12-28:
When it prints a layer of infill, doesn't it slow down to print a "bottom" layer above it if the "bottom layer speed" is slower than the rest of the print speed?  On a large area print, this can be significant, especially if there are multiple instances.

I guess I can work around it by throwing in a support blocker (worst name ever) and changing settings.

Ghostkeeper, 2020-12-28:
It prints a "bottom" layer if there is air within 1mm (or whatever the Bottom Thickness is) beneath. Similarly it prints a "top" layer if there is air within 1mm above. Anything that is not bottom or top, becomes infill. That's basically how it works right now (simplified to some extent).

Maybe I'm just visualising it wrongly though. I'm currently visualising this:
[image]

Where you're proposing to replace that one layer of infill with more 100% density skin.

The first layer of skin above infill is not printed more slowly than other skin layers.

KenLucke, 2020-12-28:
> The first layer of skin above infill is not printed more slowly than other skin layers.

OK, must be something else causing it. (Bridging, maybe?).  I'll do some snooping.  As the issue doesn't appear to be as I thought, I'll close this now.

But, when II figure it out, `<Arnold Voice>`"I'll be back"`</voice>` [grin]

closed 2020-12-28 by KenLucke