Ultimaker/Cura #8992: post processing filament change fail

azraele100877, 2020-12-21:
**Application version**
Cura 4.8

**Platform**
windows 10 home, build 19041.685' Intel(R) Core(TM) i5-5200U CPU @ 2.20 GHz, RAM 12 GB

**Printer**
Geeetech A20M version motherboard 3.1

**Reproduction steps**
1. selected filament change at the 8th layer of a print and at the 13th layer, Cura stopped and freezes after the first filament change
2. closed Cura, restarted many times and it didn't want to recognize my printer then closed everything, restarted Windows, and opened Cura, it finally started after 6 minutes of loading machines (an eternity) and recognized finally my printer.
**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2020-12-22:
We don't have a Greetech printer, so there is little we can do to debug this issue for you. You could try and add the log files of Cura, since there might be something in there, but I wouldn't get your hopes up.

azraele100877, 2020-12-23:
[cura.log](https://github.com/Ultimaker/Cura/files/5734522/cura.log)

azraele100877, 2020-12-23:
thanks @nallath , i hope that i can resolve the issue, here it is my actual log, I haven't tried again 2 colors because of the last fail, sometimes the printer stops, I am currently making some other prints now without shouting it down since yesterday and seems to work, but I am a bit afraid of stopping it!!

Ghostkeeper, 2020-12-24:
I'm supposing that this happens while printing via USB cable, right? Since you mention that Cura freezes while the printer is printing. If Cura is not connected to the printer it wouldn't have any reason to freeze while the printer is doing something particular. The log file doesn't mention anything about printing via USB. I can see that you switched to the monitor page though.

I must say that I haven't printed anything via USB in a long time. I don't have access to a USB-capable printer any more since I started working from home due to the pandemic.

Ghostkeeper, 2020-12-24:
To Cura's front-end (the only part running after slicing is completed) g-code itself is meaningless. There is no difference in sending the filament change over vs. other g-codes. It might be though that it sends temperature probes over and the printer is not responding to them during the filament change, and this would freeze up Cura? This is all done in a separate thread though so I don't understand how that could happen.

azraele100877, 2020-12-24:
You are right, I am printing via USB cable, don't know why this happens, until Cura 6.2 never had a problem, Cura 4.7 had the zits problem on the prints, cura 4.8 runs smoothly, but this error occurred...I will make another try using sd instead!

icerabbitje, 2021-05-30:
I have the problem with bad filament change post script.

Creality Ender 3.
Post-Processing to Switch Filament.

The printer completes the layers as desired. 
Then moves to a coordinate over the print and purges a blob of filament onto the work piece.
Then moves to the corner on the build plate (10,1) and starts another filament push.
The printer is grinding loudly in the process ... because the head cannot print as fast as the line is asked to move.
The printer gives off an audible alarm.
Nothing responds (controller or octoprint) and the printer needs a hard shutdown to make it stop. 

Primary issue is that the printer stops unrequested over the printed item, pushes filament instead of retracting some, then moves away to the corner and again instead of of the big retraction, pushes to seemingly purge ...

Ultimaker Cura 4.8 and 4.9.1
Latest OctoPrint on Raspberry Pi3
USB to Creality Ender 3

Ghostkeeper, 2021-06-08:
In order to fix this, we need to know what g-code is not supported by your printer that causes this. We don't have your printer, so we can't see what's going wrong. The only thing we have control over is the g-codes that Cura generates.

GregValiant, 2025-07-03:
I'll go ahead and close this as USB printing is not updated and is not recommended.
As far as the M600 goes, the implementation of the command within the Firmware will dictate how the gcode command must be configured.  There is no way to know that unless the user happens to know.

I'll go ahead and close this.  If it is still an important issue it can be re-opened.