Ultimaker/Cura #8993: [4.8.0] Text / characters are jumbled on screen during model translations

gr3333, 2020-12-21:
**Application version**
4.8.0

**Platform**
Mac. OS 11.1 on new M1 Mac Mini.

**Printer**
CR-10S

**Reproduction steps**
1. Load an STL model
2. Select the model on the build plate and attempt to move or rotate it

**Screenshot(s)**
[image]

**Actual results**
The text that shows the move/rotation is garbled/not legible

**Expected results**
I should get a legible display to assist with move

**Project file**
n/a

**Log file**
n/a

**Additional information**
As noted above, this may have to do with the graphics of the new M1 Macs.

Ghostkeeper, 2020-12-24:
This is an upstream bug in the Qt GUI framework we're using: https://bugreports.qt.io/browse/QTBUG-89379 .

It's not really anything we can do anything about until Qt fixes the problem. We can't claim to support Apple M1 at the moment because Qt doesn't support it.

jottr, 2021-11-24:
This is happening to me on Macos since `Mavericks` I belive. 
Currently I'm on `Big Sur` and the issue remains:

```
$sw_vers
ProductName:     macOS
ProductVersion: 11.6
BuildVersion:   20G165
```

`Cura v4.10.0`.

jottr, 2021-11-24:
It seems a [bugfix](https://codereview.qt-project.org/c/qt/qtbase/+/354944) has been merged to the following QT versions: `6.2 6.1 5.15 5.12`.

julianCast, 2022-08-09:
This is happening to us on a Intel chip (not a M1) but only on Big Sur OS.

GregValiant, 2024-11-20:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

gr3333, 2024-11-20:
I haven’t seen this issue recently.

closed 2024-11-20 by GregValiant