Ultimaker/Cura #8994: [4.8.0] Nozzle Change does not impact Wall Thickness

XBrav, 2020-12-22:
**Application version**
4.8.0

**Platform**
Windows 10 x64

**Printer**
Custom FFF Printer

**Reproduction steps**
1) Modify nozzle size for machine in Machine Settings (ie 0.4 to 0.6, 0.8, etc).
2) Note that all width settings update except for Wall Thickness.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Wall thickness field remains at default (0.8 aka 2 walls on a 0.4mm nozzle).
[image]

**Expected results**
Wall Thickness should adjust on nozzle size change using the dependent calculation Wall Line Count. The engine calculates Wall Line Count using the defined width / nozzle size. Ideally, this relationship should be inverted where the wall thickness is calculated using the wall counts * nozzle size.

**Project file**
[xyzCalibration_cube.zip](https://github.com/Ultimaker/Cura/files/5727283/xyzCalibration_cube.zip)

**Additional information**
(Extra information relevant to the issue.)

Asterchades, 2020-12-22:
If the Wall Thickness is set, it calculates the number of lines based on that thickness (rounding up). If you set the Wall Line Count instead it will grey out the Wall Thickness setting and calculate it based on the number of lines you've chosen.

Two different ways to describe the same basic metric, but each has its own merits.

XBrav, 2020-12-22:
> If the Wall Thickness is set, it calculates the number of lines based on that thickness (rounding up). If you set the Wall Line Count instead it will grey out the Wall Thickness setting and [...]

Either way works. The issue I've raised is that changing the nozzle size in the machine settings doesn't update either field. Most of the line width fields are dynamicly updated when a nozzle size is changed. When you change the size, it recalculates the number of perimeter fields, rather than recalculate the width based on the previous number of walls selected.

In the above situation, changing the default 0.4 nozzle to 0.6 results in the line width of 1 (0.8/0.6 rounded down to 1). Ideally, the calculation should be reversed and update the wall width as a number of x nozzle widths.

Asterchades, 2020-12-22:
I'm not sure why it would change that value with the nozzle. Line widths are tied to the nozzle size, because changing the nozzle changes the optimal width (debate, of course, rages of what the "optimal" width actually is). But the Wall Width is independent, being that you can set it to any value on any nozzle size and it won't have a direct impact on its actual ability to be printed.

I know the profile I've created myself has a default wall width based on the line width, but the normal profiles just have a fixed wall width as it doesn't actually have to change with a different nozzle.

nallath, 2020-12-22:
The value of wall_thickness is  `wall_line_width_0 if magic_spiralize else 0.8` by default. 

Wall line count is set to `1 if magic_spiralize else max(1, round((wall_thickness - wall_line_width_0) / wall_line_width_x) + 1) if wall_thickness != 0 else 0`

Note that wall thickness is only there for convenience. The engine will _only_ look at wall line count (which is also why the wall thickness is disabled when you override the wall line count!)

We could update the wall thickness calculation to `wall_line_width_0 if magic_spiralize else machine_nozzle_size * 2`, but this could change settings for other people as well.

XBrav, 2020-12-22:
> The value of wall_thickness is `wall_line_width_0 if magic_spiralize else 0.8` by default.
>
> Wall line count is set to `1 if magic_spiralize else max(1, round((wall_thickness - wall_line_width_0) / [...]

To avoid modifying the calculation, could we not add a line to recalculate the line width on nozzle size change? There already seems to be an event to update all the line widths when the field changes. Then, let the calculation recalculate wall_line_width_x.

Ghostkeeper, 2020-12-24:
If the formula for line width depends on `machine_nozzle_size`, it will automatically recalculate the line width on nozzle size change. By default the line width is the simple formula `machine_nozzle_size` so yes, it should already recalculate automatically unless your profiles override that. In your project file there is nothing that overrides that behaviour so indeed if I change the nozzle size to 0.8, the line width becomes 0.8 as well.

If you want the walls to have a fixed number of lines, override the Wall Line Count setting. It's as simple as that.

However by default we don't like to do this, because as we found, if you have two wall lines of 0.8mm that results in a similar strength as four wall lines of 0.4mm. So if you're using a bigger wall line width you can use fewer lines to achieve the same strength. Using fewer lines saves time, which is what people usually want when they use bigger nozzle sizes. We're not likely to change those defaults. You can change the settings yourself though.