Ultimaker/Cura #8999: 4.8.0 Extruder temp wrong and start script not included when wire printing enabled

SmileyNeil, 2020-12-22:
**Application version**
4.8.0

**Platform**
Windows 10 Home build 19042.685.  Nvidia GeForce RTX 2060 Super.

**Printer**
Creality CR-X (custom profile)

**Reproduction steps**
With gcode in printer's Extruder start script:
1. Set Material->Printing Temperature (and Printing Temperature Initial Layer) to 200degC.
2. Enable Experimental->Wire Printing
2. Slice

**Screenshot(s)**
With wire printing disabled (for reference):
wirePrintingDisabled.gcode
SlicedGcode_WirePrintingDisabled.jpg

With wire printing enabled (showing issue):
wirePrintingEnabled.gcode
SlicedGcode_WirePrintingEnabledStart.jpg
SlicedGcode_WirePrintingEnabledEnd.jpg

Files are at:
https://wetransfer.com/downloads/4a0a47382adfb5b5bb1e71ee5a603ca920201222183551/0454b0ee6f778fa642d6d0ce8199e33320201222183614/f4830c

**Actual results**
With wire printing enabled:
- The extruder is not set before the material temp is set.
- The material temperature is set to 210deg, irrespective of the material setting (I tried 200deg and 230deg).
- The extruder start gcode is not inserted.
- At the end of the print the printer end gcode is inserted twice, with a zero bed and extruder temp in between.

**Expected results**
- The appropriate extruder should be set before the material temperature is set.
- Requested materal temperaure should be set, instead of fixed 210degs.
- The extruder start gcode should be inserted after the printer start gcode.
- At the end of the print, the printer end gcode should only be inserted once.

**Project file**
wirePrintingBug.3mf

**Log file**
cura.log

**Additional information**
Files are at:
https://wetransfer.com/downloads/4a0a47382adfb5b5bb1e71ee5a603ca920201222183551/0454b0ee6f778fa642d6d0ce8199e33320201222183614/f4830c

Ghostkeeper, 2020-12-24:
The wire printing is kind of a completely separate code path. These bugs don't surprise me in the least.

We've actually considered removing it completely because it's not working very reliably and causing a disproportional amount of maintenance for the usage it gets. So I don't know how much chance this will have to get fixed. The problems are pretty serious though, and it hasn't been removed yet. So we'll see what my colleagues think after their Christmas holidays.

Ghostkeeper, 2021-01-04:
Since wall overlap compensation is removed in Ultimaker/CuraEngine#1210 we've decided that we'll not fix this issue. The issue will fix itself when that branch is merged.

I'll leave this open as a "known issue" until then.

GregValiant, 2024-11-19:
Wiring printing has been removed from Cura.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

closed 2024-12-08 by github-actions[bot]