Ultimaker/Cura #9000: Changing a profile will erase current profile setting

perfoWorks, 2020-12-22:
Application version
4.8.0

Platform
OSX 11.0.1 (Big Sur)

Printer
Repetier

Reproduction steps
1/ Select one custom profile
2/ Do some changes
3/ Switch to another custom profile
4/ Press discard changes to load selected profile (not exactly sure why is this even asking)
5/ Then go back to previous profile
6/ and changes are gone from step 2 which you were working on it all day

Expected results
When switching from one profile to another the settings from one profile should be automatically saved and setting from selected profile loaded. 

Project file
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

Log file
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

Additional information
(Extra information relevant to the issue.)

rburema, 2020-12-23:
>  Press discard changes to load selected profile

You literally instructed it to throw away your changes with that step. That's also why it's asking. People don't like it when their settings disappear without asking just because they wen from 0.1 to 0.15 layer height.

If you want it saved, before you change profiles (or if you dismiss the window with X) you can use 'Create profile from current settings/overrides', and if you already have a Custom profile, 'Update profile with current settings/overrides' from the profile drop-down menu.

perfoWorks, 2020-12-23:
Ok... Thanks for pointing on the button **'Update profile with current setting/overrides'** I had many profiles there for different material and different height so it is obvious that I needed to save setting in each profile before switching to another profile. I think it works with that button but I believe hiding such a important button in a dropdown menu is not very clever UI idea... Any chance to put it for example next to Recommended button? 
I'm new to Cura but I work with computers on daily basis and I didn't noticed that button and lose all my settings in 2 days of working with it so there definitely will be another new people who will pulling their hair because of this UI flaw
Thanks

Ghostkeeper, 2020-12-24:
It's putting a dialogue in your face when you change profiles that asks you whether you want to erase your current print settings or want to keep them, overlaying your changes on top of the new profile.
You said you clicked on "discard" and weren't sure why it's even asking this.

There is a problem with the user experience then. That pop-up is very prominently asking you whether you want to throw away your settings and you clicked to throw them away confidently. So perhaps you are not understanding the pop-up? Is that the problem?

Different users have different use cases for their profiles, so we're opting for not making custom profiles depend on the material. This would require people to re-make their profiles over and over for different materials which is also destructive of their work.

perfoWorks, 2020-12-24:
@Ghostkeeper I have played with that popup window few times and I know when I pressed keep changes then it saved the setting but also transferred those settings from current printing profile to selected profile.. That is not what I wanted to do. For example if I'm on ABS profile and select PLA profile then by choosing keep changes I've got all ABS settings in my PLA profile.. Therefore I have chosen discard these changes so the PLA profile stays intact but that didn't save the APS profile as mentioned above... And that's exactly poor user experience what I was talking about.. I'm probably missing something because I'm new to Cura but this is why I don't understand a reason of that popup. Why would anyone want to save and transfer setting from current profile to selected profile or discard current profile when choosing different profile.. Please fix me if I'm wrong... Sorry.. I'm not trying to be disrespectful to you. I love Cura but the user experience in some cases are really poor.. And I've heard that also from other users with years of experience in Cura. Please listen to them.
Thank you and merry Christmas

Ghostkeeper, 2020-12-28:
I think then it's best to ask a bit more towards what you'd expect the behaviour to be. This is what you wrote:

> When switching from one profile to another the settings from one profile should be automatically saved and setting from selected profile loaded.

Reading carefully though this sentence, I think you mean to say that it should first update any custom profile with the unsaved setting changes, then change the profiles. But I wonder then what you'd expect to happen if there is no custom profile, i.e. you're using a built-in profile.

> Also in perfect world each custom print setting profile should be linked to specific material. There is no point to have print settings for PLA visible when we have chosen ABS for example

They are currently tied to quality levels. There are many use cases for custom profiles and I think your request would tune it too far for your specific use case. Some people adjust profiles to work well with their application, i.e. a profile for strength that uses the Cubic Subdivision infill pattern and high infill density. Or they create profiles for specific printers because they know that this one printer has a rather worn-down feeder wheel and should retract more carefully. These things are not specific to materials, so for them it would be annoying to have to re-create their profile for every material they're using.

perfoWorks, 2021-01-02:
@Ghostkeeper 

> Reading carefully though this sentence, I think you mean to say that it should first update any custom profile with the unsaved setting changes, then change the profiles. But I wonder then what you'd expect to happen if there is no custom profile, i.e. you're using a built-in profile.

I believe that build in profile should not be editable.. This is only my opinion.. I believe that build in profile should be there as recommended setting and if someone would want to adjust them then I believe that this build in profile should be duplicated which would create a new custom profile which then could be modified to their needs. 

> They are currently tied to quality levels. There are many use cases for custom profiles and I think your request would tune it too far for your specific use case. Some people adjust profiles to work [...]

I believe link each custom profile to a printer and material would make it much easier to adjust something and organise each custom profile.. If someone switch from one printer to another and would want to use profile from another printer and another filament manufacture then it would be nice to just add the link to that custom profile they want to use or once again duplicate that profile and adjust it. 

This is only my opinion and I know there's more into it. All I know that transferring setting from one profile to another when pressing keep changes in popup window or discarding all changes when pressing discard button is just wrong. 

Thanks for taking time with it

Ghostkeeper, 2021-01-11:
I think it'd also be a bit strange if you change material that it pops up with a question what you'd like to name your new profile though.

Changing custom profiles to be material-specific is pretty much out of the question for Cura as it stands now.

* This would mean that existing custom profiles from 4.8 and earlier would disappear completely.
* If a custom profile is based on a quality level that isn't supported for the user's new nozzle, but is supported for the material, you'd end up with a broken stack or at least being unable to select that profile.
* Some printers don't have materials as a configuration option.
* It breaks the workflow for users that design custom profiles for a specific intent, e.g. a high-strength profile. This is one of the main use cases for profiles right now.

The pop-up can remember your choice. Personally I just have it set to "keep changes" all the time, because I also don't like it discarding my changes automatically. (I'll discard it manually if necessary.) But from user tests this option was seen as a necessity.

lvitol, 2021-01-13:
> overlaying your changes on top of the new profile

I am new to cura, this confuse me a lot，I'm not expecting the changes to overlay the new profile.

fvrmr, 2021-01-19:
> I am new to cura, this confuse me a lot，I'm not expecting the changes to overlay the new profile.

If you are new to Cura the plugin Settings Guide can help you, you can find that plugin in the Marketplace. 
You can find more information about profile settings here: https://support.ultimaker.com/hc/en-us/articles/360012381720-Switching-print-profiles-with-unsaved-changes
To keep this issue clean I will mark your comment as off topic.

gr3333, 2021-02-22:
> It's putting a dialogue in your face when you change profiles that asks you whether you want to erase your current print settings or want to keep them, overlaying your changes on top of the new [...]

I too have had this issue, and looked for a way to save 'my settings' options and never found it (under main menu options).
Would it be possible to provide a 'save' button or a hint to look under the pop up?