Ultimaker/Cura #9001: add selection tools and apply settings to specific parts of the model based on selection

moham96, 2020-12-23:
**Is your feature request related to a problem? Please describe.**
sometimes I want to apply specific settings(infill, support, speed, cooling) to specific areas of the model to achieve the best quality, strength, or flexibility. unfortunately, Cura doesn't provide such tools.
some examples:
 - Use 20% infill in the model but I want to use 70% or more infill just around screws for strength.
 - decrease infill at some places where I want the part to flex but keep it high at the rest of the model for strength.
 - decrease speed in some places where I want to get the best details without increasing time drastically by increasing the whole model speed.
 - increase cooling(and maybe decrease speed also) in places where I want the overhangs to be printed without support, I know I can adjust those based on overhang angle but being able to set these at specific overhangs gives you so much power on optimizing for time and material waste.

**Describe the solution you'd like**
My suggestion is to have selection tools to select the part of the model and then apply specific settings to those selections.
for example, the "support blocker" should be turned to a "selection blocker" where you place it in the model and then the intersection of the model and the blocker can have specific settings (no support in this example) but you can apply other settings too. 
Maybe add another selection tool where instead of a cube blocker we have a cylinder (to apply settings around screws for example),
Another selection tool could be the brush tool similar to the one added recently to set custom support in `Prusa slicer`.
another one is the selection tool in mesh mixer where you can easily select lines/parts of the model or deselect them.

nallath, 2020-12-23:
Cura has the option to do all of the things you suggest. You can find more info about it here: https://support.ultimaker.com/hc/en-us/articles/360013370140-How-to-adjust-print-settings-of-a-part-of-my-model-in-Ultimaker-Cura

closed 2020-12-23 by nallath

Ghostkeeper, 2020-12-24:
[image]

nallath, 2020-12-24:
Doh. I can't believe I missed that. Glad you caught it @Ghostkeeper :D