Ultimaker/Cura #9003: "Support Block" option Missing on sidebar in Cura Interface

Redspectre, 2020-12-23:
**Cura 4.8.0**
(The version of the application this issue occurs with.)

**Windows 10**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Creality ender 3**
(Which printer was selected in Cura?)

**Reproduction steps**
Installed cura from site

**Screenshot(s)**
[image]

(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
I should be able to see the support block icon on the sidebar
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-12-23:
Your support is missing a clear description of what is going wrong (as well as many other bits we would need to diagnose the problem), but the Support Blocker is showing just right in the toolbar. It is the 6th icon.

The toolbar is supposed to have tooltips. These will return in the next major version of Cura.

Redspectre, 2020-12-23:
Forgive my clumsiness, But I meant this icon, (thats usually seen at the end of the sidebar that lets us create new shapes)

[image]

fieldOfView, 2020-12-27:
The icon changed. The tool is the same.

Ghostkeeper, 2020-12-28:
Yup, the new icon is this:
[image]

The tooltips will be fixed in the next release (thanks to fieldOfView) so this should be a bit more clear then.

closed 2020-12-28 by Ghostkeeper

Redspectre, 2020-12-29:
But it is not letting me add in any new shape.

[image]

Ghostkeeper, 2021-01-06:
What happens when you click your flower model while that tool is active?

Redspectre, 2021-01-11:
Nothinh happens when I click It, it just stays highlighted, no popups / new tabs opening to create bodies

fvrmr, 2021-01-18:
Did you click on your model? Because this should happen:

https://user-images.githubusercontent.com/77000924/104927646-f9c55d00-59a1-11eb-849c-8eb6b9f93b5b.mp4

Redspectre, 2021-01-19:
yes, I get the shadowy cube, But how do i change the sizes or geometry of the cube?
 Am trying to import another profiled geometry in the main part and set different infill properties.

fvrmr, 2021-01-19:
You can click on the shadowy cube and resize and place it where you want!

Redspectre, 2021-01-21:
> You can click on the shadowy cube and resize and place it where you want!

Ah Thanks, I got to change to sizes, 

is there any way to input custom geometry and set support block?

fieldOfView, 2021-01-21:
Yes; load the custom geometry and use the "Per Model Settings" tool to make it "Print as support". The Per Model Settings tool is the second from the bottom (ie the one with the 4 objects in it).

Redspectre, 2021-01-28:
Thank you all.