Ultimaker/Cura #9004: 999999 bottom layers

hygroscopiccarpaccio, 2020-12-23:
**Application version**
(The version of the application this issue occurs with.)
4.8.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Linux mint 20, x86_64, 5.4.0-58-generic
intel i5-10300H

**Printer**
(Which printer was selected in Cura?)
Wanhao i3

**Reproduction steps**
1. (Something you did.)
printing something with an infill of 100%

2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

The calculated bottom layers changes from 8 to 999999 and the top layer changes from 8 to 0
Changing the infill to 99.99% returns the bottom and top layers to 8

**Expected results**
(What should happen after the above steps have been followed.)
Dont change the bottom my top and bottom layers to 999999

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
No errors are reported on this........  but below is a snippet

2020-12-23 17:29:30,679 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G1 F200 E6 ;extrude 6 mm of feed stock
2020-12-23 17:29:30,685 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G92 E0 ;zero the extruded length again
2020-12-23 17:29:30,690 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] G1 F120
2020-12-23 17:29:30,700 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] ;Put printing message on LCD screen
2020-12-23 17:29:30,706 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [110]: [Backend] M117 Printing..." -s material_flow_dependent_temperature="False" -s retraction_hop="1" -s dual="0" -s z_seam_relative="False" -s support_roof_angles="[ ]" -s layer_height_0="0.3" -s material_anti_ooze_retracted_position="-4" -s ironing_pattern="zigzag" -s machine_heated_build_volume="False" -s minimum_bottom_area="1.0" -s meshfix_keep_open_polygons="False" -s wireframe_printspeed="5" -s cutting_mesh="False" -s acceleration_support_infill="3000" -s infill_pattern="grid" -s material_shrinkage_percentage="100.0" -s skin_line_width="0.4" -s wipe_move_distance="20" -s time="17:29:29" -s travel_compensate_overlapping_walls_x_enabled="True" -s infill_mesh_order="0" -s machine_min_cool_heat_time_window="50.0" -s cool_min_speed="10" -s support_bottom_material_flow="100" -s infill_extruder_nr="-1" -s machine_depth="200" -s wireframe_straight_before_down="20" -s material_bed_temp_wait="True" -s gradual_infill_step_height="1.5" -s support_conical_min_width="5.0" -s infill_overlap="10" -s draft_shield_enabled="False" -s material_break_temperature="50" -s wipe_hop_speed="10" -s support_bottom_pattern="concentric" -s machine_nozzle_id="unknown" -s material_bed_temperature_layer_0="80" -s roofing_layer_count="0" -s jerk_travel="30" -s material_anti_ooze_retraction_speed="5" -s material_break_preparation_speed="2" -s wireframe_flow="100" -s jerk_travel_layer_0="30.0" -s speed_roofing="30.0" -s extruder_prime_pos_z="0" -s machine_nozzle_expansion_angle="45" -s cool_fan_speed_min="100.0" -s switch_extruder_extra_prime_amount="0" -s support_conical_enabled="False" -s gradual_infill_steps="0" -s jerk_infill="20" -s wall_min_flow_retract="False" -s material_guid="" -s connect_infill_polygons="False" -s ironing_only_highest_layer="False" -s bridge_settings_enabled="False" -s mesh_position_x="0" -s skin_material_flow="100" -s wireframe_roof_inset="3" -s support_connect_zigzags="True" -s material_end_of_filament_purge_length="20" -s machine_use_extruder_offset_to_offset_coords="True" -s machine_endstop_positive_direction_x="False" -s support_material_flow="100" -s wipe_retraction_prime_speed="25" -s cross_infill_density_image="" -s support_roof_offset="0.0" -s support_skip_some_zags="False" -s bridge_fan_speed_3="0" -s wipe_repeat_count="5" -s support_fan_enable="False" -s infill_sparse_thickness="0.1" -s experimental="0" -s material_end_of_filament_purge_speed="0.5" -s prime_tower_flow="100" -s machine_max_feedrate_e="299792458000" -s top_bottom_pattern_0="lines" -s infill_angles="[ ]" -s machine_nozzle_size="0.4" -s skin_angles="[ ]" -s raft_interface_thickness="0.15000000000000002" -s nozzle_disallowed_areas="[]" -s meshfix_maximum_resolution="0.5" -s support_angle="50" -s expand_skins_expand_distance="0.8" -s material="0" -s support_roof_enable="False" -s machine_max_jerk_e="5.0" -s skin_overlap_mm="0.02" -s date="23-12-2020" -s machine_heat_zone_length="16" -s shell="0" -s travel_avoid_supports="False" -s bridge_sparse_infill_max_density="0" -s print_temperature="210" -s material_initial_print_temperature="200" -s wireframe_roof_outer_delay="0.2" -s machine_nozzle_head_distance="3" -s skirt_brim_line_width="0.4" -s machine_nozzle_temp_enabled="True" -s raft_surface_fan_speed="0" -s acceleration_print_layer_0="3000" -s limit_support_retractions="True" -s material_bed_temperature="70" -s prime_tower_min_volume="6" -s material_standby_temperature="150" -s support_roof_density="100" -s jerk_support_bottom="20" -s conical_overhang_enabled="False" -s speed_travel="120" -s support_use_towers="True" -s jerk_support_infill="20" -s wipe_retraction_enable="True" -s magic_fuzzy_skin_point_dist="0.8" -s mold_width="5" -s brim_replaces_support="True" -s speed_print="60" -s minimum_interface_area="1.0" -s infill_before_walls="True" -s connect_skin_polygons="False" -s support_z_distance="0.1" -s support_bottom_angles="[ ]" -s raft_surface_line_spacing="0.4" -s material_break_retracted_position="-50" -s wipe_retraction_extra_prime_amount="0" -s bridge_skin_density_3="80" -s cool_min_layer_time="5" -s wall_0_inset="0" -s speed_support_interface="20.0" -s ironing_inset="0.38" -s skin_preshrink="0.8" -s support_tower_diameter="3.0" -s material_print_temp_prepend="True" -s material_flow_temp_graph="[[3.5,200],[7.0,240]]" -s gantry_height="180" -s support_interface_skip_height="0.3" -s wireframe_printspeed_up="5" -s travel_avoid_other_parts="True" -s wipe_hop_amount="1" -s roofing_line_width="0.4" -s adhesion_type="skirt" -s wireframe_flow_flat="100" -s platform_adhesion="0" -s wall_line_width_x="0.4" -s raft_base_fan_speed="0" -s machine_disallowed_areas="[]" -s switch_extruder_retraction_speeds="20" -s z_seam_corner="z_seam_corner_inner" -s roofing_angles="[ ]" -s support_tree_branch_diameter="2" -s prime_tower_size="20" -s wireframe_drag_along="0.6" -s brim_line_count="20" -s skin_edge_support_thickness="0" -s travel_avoid_distance="0.625" -s material_print_temperature_layer_0="210" -s layer_0_z_overlap="0.15" -s speed_layer_0="30.0" -s infill_randomize_start_location="False" -s top_skin_expand_distance="0.8" -s slicing_tolerance="middle" -s adaptive_layer_height_threshold="0.2" -s support_roof_pattern="concentric" -s jerk_prime_tower="20" -s raft_speed="15.0" -s bridge_skin_material_flow_2="100" -s machine_endstop_positive_direction_y="False" -s ooze_shield_angle="60" -s wireframe_roof_drag_along="0.8" -s jerk_support_roof="20" -s extruders_enabled_count="1" -s extruder_prime_pos_y="0" -s retraction_min_travel="0.8" -s support_line_distance="2.6666666666666665" -s speed_print_layer_0="30.0" -s machine_shape="rectangular" -s support_interface_angles="[ ]" -s roofing_extruder_nr="-1" -s cool_fan_speed_max="100.0" -s travel_compensate_overlapping_walls_0_enabled="True" -s support_mesh="False" -s material_flow="100" -s retraction_extrusion_window="6.5" -s small_feature_max_length="0.0" -s skin_outline_count="1" -s acceleration_topbottom="3000" -s machine_gcode_flavor="RepRap (Marlin/Sprinter)" -s material_extrusion_cool_down_speed="0.7" -s material_no_load_move_factor="0.940860215" -s speed_ironing="20.0" -s skin_edge_support_layers="0" -s support="0" -s machine_extruders_share_heater="False" -s speed_support_infill="30" -s support_skip_zag_per_mm="20" -s meshfix="0" -s speed_wall_0="30.0" -s machine_always_write_active_tool="False" -s layer_start_y="0.0" -s prime_tower_wipe_enabled="True" -s wireframe_printspeed_bottom="5" -s coasting_volume="0.064" -s wall_x_extruder_nr="-1" -s raft_base_jerk="20" -s raft_surface_layers="2" -s jerk_topbottom [... cut ...]
2020-12-23 17:29:34,991 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2020-12-23 17:29:34,993 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
2020-12-23 17:29:34,993 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [898]: Backend quit with return code 0. Resetting process and socket.

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-12-27:
This is intentional.

When printing at 100% density, Cura replaces all infill with skin. This way you won't get any border between the skin and the infill any more and achieve a greater infill density. It won't need to plan the lines separately (no longer printing skin afterwards or anything) so it just continues with straight lines longer. What's more, the skin settings will have been tuned for 100% density better.

This is one of those cases where automation is very counter to intuition, and Cura chooses to work better out of the box at the cost of working intuitively or predictively.

closed 2020-12-27 by Ghostkeeper

Nur-Kekse-im-Kopf, 2020-12-30:
Hi, first of all, I'm not trying to be rude, even if it may sound like that.

I want to second the behaviour being a bug. It may be intentional, but it is still a bug (only by design then). I'll try to explain ...

There are very valid reasons to have separate infill and and top/bottom even for solid (as in 100 % infill) parts. Just to name a few examples:

# Wanting different fill patterns for top/bottom vs infill
I often arrange infill in a way that prints relatively fast and has something to do with expected loads (e.g. going for quasi isotropic with something like lines in 0/90/+45/-45...), whereas I might want an aesthetically pleasing top/bottom layer (e.g. concentric).

# Optimising print speeds

# Having control over perimeter count, overlap etc. (which is somewhat possible with infill, but not really with top/bottom*)

But even if you say "No, we know better than our users what they want and they WANT that being the default behaviour", there is still a bug in it: If you change the bottom height setting (in millimetres) manually, it has no effect (it should!), but if you change the bottom layer count setting instead, it does. WTF? At least make it so that manually setting the bottom height will affect the bottom layer count (as it should).

Nur-Kekse-im-Kopf, 2020-12-30:
Sry for the weird formatting. These should have been bullet points instead.

fieldOfView, 2020-12-30:
There is nothing stopping you from overriding the default behavior. You can set the number of bottom layers back to a "normal" amount if you want to for example use a different pattern for top/bottom vs infill.

Nur-Kekse-im-Kopf, 2020-12-30:
@fildOFView

Yes, one can manually change the bottom layer count (which I mentioned above). However, why does manually changing the bottom height (value in mm instead of integer layer number) not do the same? The latter one is a bug and should be fixed.

Ghostkeeper, 2021-01-08:
> why does manually changing the bottom height (value in mm instead of integer layer number) not do the same?

Many profiles override the Bottom Height setting, including many user profiles, without intending it to break this functionality when printing at 100% density.