Ultimaker/Cura #9005: Arachne engine alpha - won't slice Nylon & PVA

markyguk, 2020-12-23:
[image]

**Application version**
Arachne engine alpha

**Platform**
Windows 10

**Printer**
S5

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
attached

**Actual results**
unable to slice, just stalls

**Expected results**
should have sliced
[UMS5_Nutcracker_New.zip](https://github.com/Ultimaker/Cura/files/5736558/UMS5_Nutcracker_New.zip)

**Project file**
attached as zip

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
Doesn't error out, just stalls. Going to try with PLA instead, just happened to have Nylon loaded and thought i would see what would happen.

jellespijker, 2020-12-23:
Thank you for testing the Arachne alpha. Could you provide the log files as well. These will probably contain a lot of warning that can help us figure out why it crashed.

nallath, 2020-12-24:
I've just tried to slice the project, it worked for me (although I am slicing it on Linux, so it could be that it's an issue on windows only)

markyguk, 2020-12-24:
hi, so it doesn't actually crash Cura it just stops slicing. If I click cancel its fine.

I originally tried on surface pro, thought it might be an issue with that. tried again on an alienware computer and same issue. Running windows 10. It just seems to stop slicing. It doesn't crash, so doesn't create an error log. 

also I tried to slice with Cura 4.7 and that works fine. 

I tried playing around with a few options. If i have support material set to extruder 1 or no supports then it slices fine. Only seems to cause an issue when using both extruders and support material set to extruder 2.
i have tried with Nylon and with PLA in extruder 1 (AA 0.4), and PVA in extruder 2 (BB 0.4) this is all ultimaker brand filament on default profiles. 

I will try and change some of the settings and also see if the AA 0.8 & BB 0.8 extruders behave the same.

nallath, 2020-12-24:
Can you post the log files? If it stops slicing it could be because of an engine crash (We should be able to see if that's the case in the logs)

markyguk, 2020-12-24:
[cura.log](https://github.com/Ultimaker/Cura/files/5740132/cura.log)

log file attached, hopefully that helps.

please let me know if you want me to try again on different settings.

jellespijker, 2020-12-24:
The engine throws a `return code 3221225477` Most likely pointing to an edge in the graph which got delete and isn't there anymore. @markyguk Could you try rotating it 90 deg or moving it and try to slice it again?

markyguk, 2020-12-24:
that seems to have fixed it, im sure i tried that originally but anyway seems to be working now.

thanks :) 

and Happy Holidays

jellespijker, 2020-12-25:
Devs: I think this might be an other case of a deleted twin junction

closed 2024-12-08 by github-actions[bot]