Ultimaker/Cura #9010: Segmentation fault when sending job

kuon, 2020-12-24:
I use the duet plugin ( https://github.com/Kriechi/Cura-DuetRRFPlugin/issues/24 ), but when I try to send a job, I get this crash.

Cura version, master on 9efe5dd5e29f36732148731c389c8bb280ab2402

Log: https://gist.github.com/kuon/bb376768203c5ef8c42423064c7d62e2

nallath, 2020-12-24:
I do see that the plugin isn't using the HttpRequestManager, which should handle this a bit better . 

It's probably some issue with the version of Qt that we use, which for some reason doesn't pop up with our own uploading. My suggestion would be to check if the issue occurs when sending it in the same way as that we send things to the cloud.

Ghostkeeper, 2020-12-28:
Yeah so here we have a plug-in that calls upon QNetworkRequest to upload a file. Cura allows the plug-in to talk with Qt there because it's a package that contains both. However you're only getting a crash when using your distribution package, not with the AppImage.

We are in control of Cura's source code as well as how it's packaged in the AppImage. But this crash is not with Cura's source code (it's in the plug-in) nor our package. You are using Qt 5.15 through Python 3.9, but Cura **requires** CPython version 3.5.2 and Qt 5.10 exactly. This difference is probably why it's not crashing with our package but it does crash with the distribution you're using.

I don't really see how we can fix this. It's not in our zone of control. I see that the developer of that Duet plug-in referred you to us, and I hate to do it, but this is really between him and the maintainer of that Cura distribution package that you're using. Sorry!

closed 2020-12-28 by Ghostkeeper

kuon, 2020-12-31:
The problem is that the appimage version constantly crashes and is unusable under wayland.

Kriechi, 2020-12-31:
Hey - original maintainer of the plugin in question here.

So essentially we are both passing it "up the chain" and now we are looking for the fault in the distribution of the Cura package? Fascinating bug we have on our hands here 😄 

My initial claim why I referred the bug to the Cura project was simply based on the idea that I'm using Python code - and Python code should not in any way cause a segfault - hence "passing it up the chain". By the same argument of course the Cura maintainers can pass it on as well, so at least for me narrowing down the bug in the distribution makes sense - and would also explain the challenges of not being able to reproduce it.

I'm a bit confused about the statement that I would be using the wrong API... I see plenty of results when grepping for `QNetworkRequest` under `plugins/`. Are we all using the wrong API, or is there a finer point on which API should be used where (and by whom)?

I'm happy to use a more appropriate API if you think this would help - @Ghostkeeper could you point me to an example or API SDK documentation, or can I simply use one of the other existing plugins in the Cura repo that references `HttpRequestManager`?

Ghostkeeper, 2021-01-08:
Indeed, Python code should never cause a segfault, which is indeed why the bug is being passed around here ;) It's really a bug in the version of Qt that Kuon is using, like Nallath said. It's apparently not a bug yet in the version that Cura is supposed to be using.

The code you are using should work and apparently does work in the official distribution. We are only required to support our own distribution here. We don't have time to support others. That's why we are quick to close this issue.

You'll find references to `QNetworkRequest` because `HttpRequestManager` gives that as parameter to its callbacks. It's used as typing. There is still one place though that still uses `QNetworkRequest`: The UM3 Network Connection plug-in and its friends in the networked printer output. That part still needs to be refactored by us, and it's also the most complex part.

The `HttpRequestManager` internally also uses `QNetworkRequest`, but hides that from the API and handles the request safely with how it's constructed and deleted (quite an issue when dealing with C++ objects in Python that get sent over multiple threads in callbacks like this). It could still crash, but since other parts of Cura apparently don't cause a crash (e.g. when checking for new releases of Cura at start-up, if the preference is enabled) there's a good chance that the `HttpRequestManager` works around the bug in Qt.

Like most things in Cura we don't have separate SDK documentation pages but you can generate them using Doxygen on the Cura or Uranium repository.
Myself I always just read them from within the code, but that's just me. The class is here:
https://github.com/Ultimaker/Uranium/blob/master/UM/TaskManagement/HttpRequestManager.py

You'll probably just want something like:

    HttpRequestManager.getInstance().post("http://path/to/printer", headers, printer_data, completed_callback)

Ghostkeeper, 2021-01-08:
> The problem is that the appimage version constantly crashes and is unusable under wayland.

Probably want to submit that as a separate issue then. Although we did get a theme fix that might affect/fix this issue in the next release: https://github.com/Ultimaker/Cura/pull/7139

kuon, 2021-01-08:
> > The problem is that the appimage version constantly crashes and is unusable under wayland.
> 
> Probably want to submit that as a separate issue then. Although we did get a theme fix that might affect/fix this issue in the next release: #7139

Yeah, I should do that, but I realize wayland is still unstable on many level, and I don't want to open issues everywhere when wayland still has a long way to go.