Ultimaker/Cura #9011: Cura crashes after MacOS Big Sur update

MichalPlays, 2020-12-24:
**Application version**
4.8

**Platform**
MacOS Big Sur 11.1 / AMD Radeon Pro 5500M 4GB

**Printer**
Creality Ender 3 Pro

**Reproduction steps**
1. Updating to latest MacOS (From High Sierra to Big Sur)
2. Attempting to start the program after an successful OSX update 

**Actual results**
I click on the icon and it doesn't launch. I try to reinstall Cura abd this also doesn't seem to work

**Expected results**
App launches without any issues and I'm able to work on my files.

**Log file**
2020-12-24 08:56:30,520 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FileLogger
2020-12-24 08:56:30,526 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin LegacyProfileReader
2020-12-24 08:56:30,531 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin AMFReader
2020-12-24 08:56:30,536 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SimpleView
2020-12-24 08:56:30,543 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SolidView
2020-12-24 08:56:30,551 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin FirmwareUpdateChecker
2020-12-24 08:56:30,556 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SentryLogger
2020-12-24 08:56:30,562 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin SupportEraser
2020-12-24 08:56:30,567 - INFO - [MainThread] UM.Logger.info [122]: Loaded plugin PrepareStage
2020-12-24 08:56:30,569 - DEBUG - [MainThread] UM.PluginRegistry.loadPlugins [432]: Loading all plugins took 0.8112599849700928 seconds
2020-12-24 08:56:30,574 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [163]: Looking for old configuration files to upgrade.
2020-12-24 08:56:30,587 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,589 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,592 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,595 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,598 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,600 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,603 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,605 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,607 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,610 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,612 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,614 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,616 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,619 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,621 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,624 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,626 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [332]: Failed to get file version: ./.DS_Store, skip it
2020-12-24 08:56:30,633 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [175]: Checking and performing updates took 0.05855703353881836
2020-12-24 08:56:31,612 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.loadAllMetadata [361]: Loading metadata into container registry took 0.7817490100860596 seconds
2020-12-24 08:56:31,623 - DEBUG - [MainThread] cura.Scene.CuraSceneController.setActiveBuildPlate [142]: Select build plate: 0
2020-12-24 08:56:31,625 - INFO - [MainThread] cura.CuraApplication.run [782]: Initializing machine error checker
2020-12-24 08:56:31,944 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition creality_ender3 failed to pickle.
2020-12-24 08:56:31,957 - INFO - [MainThread] cura.UI.MachineActionManager.addDefaultMachineActions [74]: Default machine actions added for machine definition [creality_ender3]
2020-12-24 08:56:32,224 - WARNING - [MainThread] LocalContainerProvider.LocalContainerProvider._saveCachedDefinition [295]: The definition cache for definition creality_base failed to pickle.
2020-12-24 08:56:32,268 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [194]: Extruder[creality_base_extruder_0 #2] added to [Creality Ender-3] at position [0]
2020-12-24 08:56:32,269 - INFO - [MainThread] FirmwareUpdateChecker.FirmwareUpdateChecker.checkFirmwareVersion [79]: No machine with name Creality Ender-3 in list of firmware to check.
2020-12-24 08:56:32,828 - DEBUG - [MainThread] cura.Machines.ContainerTree.__getitem__ [120]: Adding container tree for creality_ender3 took 0.553386926651001 seconds.
2020-12-24 08:56:32,836 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: ChangeAtZ
2020-12-24 08:56:32,838 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: ColorMix
2020-12-24 08:56:32,839 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: DisplayFilenameAndLayerOnLCD
2020-12-24 08:56:32,840 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: DisplayProgressOnLCD
2020-12-24 08:56:32,842 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: FilamentChange
2020-12-24 08:56:32,843 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: InsertAtLayerChange
2020-12-24 08:56:32,846 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: PauseAtHeight
2020-12-24 08:56:32,847 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: RetractContinue
2020-12-24 08:56:32,848 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: SearchAndReplace
2020-12-24 08:56:32,852 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: Stretch
2020-12-24 08:56:32,853 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: TimeLapse
2020-12-24 08:56:32,854 - DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [202]: Begin loading of script: UsePreviousProbeMeasurements
[... 40 lines cut ...]
2020-12-24 08:56:35,218 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/composite.shader]...
2020-12-24 08:56:35,225 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/overhang.shader]...
2020-12-24 08:56:35,230 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/striped.shader]...
2020-12-24 08:56:35,235 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/transparent_object.shader]...
[... cut ...]
2020-12-24 08:56:42,045 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [315]: Attempting to kill the engine process
2020-12-24 08:56:42,045 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [321]: Killing engine process
2020-12-24 08:56:42,046 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [324]: Engine process is killed. Received return code -15

**Additional information**
crash message [https://pastebin.com/QzyVmAfn](url)

Would be really happy if someone could help me with this issue

tubaian, 2020-12-24:
Same problem here..

Ghostkeeper, 2020-12-28:
In that crash log it indicates that CPython crashes before it's even executed a single line of Cura's own code. Must be a bug in CPython or its packaging then.

MichalPlays, 2021-01-01:
Thank you for the reply and happy new year.
I tried to use a community made command line launcher from the brew repository and disabled that CPython thing. 
Well, it looks like this solved the first issue but now I cannot connect to my printer.
But hey, at least it launches, the main features work and I can just send the stuff from my very old Laptop that runs with Linux and even thought Cura freezes for a sec. each time I press something on the Ui, I am still able to sent the project to my printer.

> In that crash log it indicates that CPython crashes before it's even executed a single line of Cura's own code. Must be a bug in CPython or its packaging then.

closed 2021-01-01 by MichalPlays

fieldOfView, 2021-01-03:
> I tried to use a community made command line launcher from the brew repository and disabled that CPython thing.

I'm sure more information on that workaround would by appreciated.