Ultimaker/Cura #9015: Option to heat bed and hotend same time

mrx23dot, 2020-12-24:
Please add option to start heating up the **bed** and **hotend** at the same time.

This would save a few minutes in every print.

I know some Ultimaker can't handle this, but most other printers can. 
(during normal printing we can have both on at the same time because they are independent)

nallath, 2020-12-24:
You can just set this in the start g-code of your printer. 

Something like should do the trick:
```
M140 S{material_bed_temperature} ; Set bed temperature, but don't wait
M109 S{material_print_temperature} ; Set nozzle temperature and wait
M190 S{material_bed_temperature} ; Wait until the bed has reached the target temp
```

closed 2020-12-24 by nallath

mrx23dot, 2020-12-27:
Interesting, the generated heating rutin comes **before** the custom printer specific start code.
So it makes my extra heating code ineffective. (I also added some Z raise to protect the bed)

I don't think it was meant to work this way.

fieldOfView, 2020-12-27:
Have you _tried_ what @nallath suggested?

Cura checks if your start gcode snippet contains heating statements by looking for texts like `{material_bed_temperature}` and `{material_print_temperature}` respectively. If your start gcode snippet does not, Cura protects you from cold extrusion by adding its own heating sequence before your start gcode snippet. However if your custom start gcode snippet does contain a correct heating sequence, then Cura does not add its own before your start gcode.

fieldOfView, 2020-12-27:
Use the actual text `{material_print_temperature}`, don't fill in numbers.

mrx23dot, 2020-12-28:
Thank you, I needed to set it in 2 steps, otherwise hotend oozes while bed heats.
Also needed to use initial temps.

material_print_temperature_layer_0
material_bed_temperature_layer_0