Ultimaker/Cura #9016: [4.8.0] Shared Heater setting missing in version 4.8.0

seantapscott, 2020-12-24:
**Application version**
4.8.0

**Platform**
Windows 10

**Printer**
All printers missing the Shared Heater setting in version 4.8.0.  This Printer Settings option was available in 4.6.2 . 

**Reproduction steps**
Try printing with a shared heater with dual extruders.  The second heater is never told to heat up because 4.8.0 has forgotten to include that setting.

seantapscott, 2020-12-24:
[image]
[image]

GregValiant, 2020-12-25:
In the Marketplace, under Plugins, download Printer Settings.  It's in there now.  Number 16 in the list.

seantapscott, 2020-12-25:
This does not appear to be a real solution.  According to FieldOfView's own Readme file for this plugin, this new Printer Settings panel only affects the configuration files that normally have to be edited manually. 

It doesn't affect the fact that 4.8.0 is not generated gcode properly for a shared heater printer, and 4.6.2 did.  This is a regression testing problem, and a dropped feature between versions.

GregValiant, 2020-12-26:
That is odd.  Using 4.8.0 - selecting the "Extruders Share Heater" option in "Printer Settings" generates proper Gcode for my 2-in-1 out hot end.  It reacts correctly when I set the extruders to different print temperatures.  It does makes those temperature changes a few lines of Gcode before the actual tool change.  If they used M109 S and R it would probably just cause oozing during the wait.  I just switched from the beta version of 4.8 but it worked in the beta as well.  Having to search around and load a plugin to access the setting was bit annoying.

seantapscott, 2020-12-26:
I've got a SeeMeCNC rostock Max v4 with the y-adapter (not that it is terribly relevant), and I'm printing a simple two-color model with an extruder filament switch at the start of a layer.  The gcode flavor is RepRap, but this may have been sliced with Marlin flavor. 

I'm wondering if this is a RepRap vs Marlin flavor issue.  

Despite that, here is the filament change section, and there is no warmup created. I'm not entirely sure what the edge case is here. 
[image]

GregValiant, 2020-12-27:
Marlin uses M84 to disable steppers and RepRap uses M18 so I'd say it's a RepRap gcode.  The first line of the Gcode file should give the flavor. 
If both your materials use the same printing temperature there wouldn't need to be a temperature change or warmup around the tool change.  There will be an M104 and M109 at the beginning of the file and an M104 S0 at the end of the file.
If the printing temps are different for extruder 0 than extruder 1 then somewhere above the tool change will be (should be) an M104 line so that when the tool change comes up the extruder is near the correct temperature.  So just looking at that snippet it looks like is was sliced for a "shared heater".  Although if it was sliced for separate heaters, and if the standby temperature was equal to the printing temperature, it might look the same.
Looking at the retractions before tool change - you have a normal retraction after extruding of 10, then a retraction of 130, then a retraction of 5, and finally a retraction of 155 for a grand total of 300mm of retraction of T0 before switching to T1.  Then T1 primes a total of 31 before starting to extrude.  I guess since there is just the single color switch it doesn't matter but that isn't even close to being even.

seantapscott, 2020-12-27:
OK.

4.8.0 dropped the Shared Extruder feature. It doesn't work with shared
heater extruders.  Glad it is working for you, not really looking to argue
about it.  I'm just here to report that the setting is missing, support for
the setting is missing, and the gcode is the same as 4.6.2 due to
regression test failure on the shared heater setting.  If it worked for
you, it is probably because the temperature for the second tool was preset
to a certain temperature, and/or you might not be using RepRap gcode or a
WifiDuet.

Ghostkeeper, 2020-12-28:
The interface element was removed from the Machine Settings dialogue in 4.7 due to it requiring additional changes to the start and end g-codes. It doesn't work as a checkmark by itself. We leave it up to the printer definition writers to set the setting if it applies to their printer, so they can then also set correct start and end g-codes. For printers that are not in the list of 300 that ship with Cura (and would need Custom FFF Printer), they would need to use the Printer Settings plug-in from the Marketplace.

It's intentional that it's no longer in that dialogue.

The setting still works the same way in 4.7+ as it did in 4.6.

For more information, see https://github.com/Ultimaker/Cura/issues/8148. I'll close this thread as a duplicate of that one. It explains the whole story again.

closed 2020-12-28 by Ghostkeeper

ArteFlux, 2021-01-28:
Hello seantapscott, 

I think you could add this to your printer profile to define it to use the shared heater option:

		"machine_extruders_share_heater": { "value": true },

seantapscott, 2021-01-28:
meh, discussed above by using the Printer Settings plugin, poor solution.  When I checked my printer configuration files, this settings was also already there, and the gcode as still being generated incorrectly.  But hey, me and every other person using a SeeMeCNC Rostock printer can't be wrong, can they?

ArteFlux, 2021-01-28:
> me and every other person using a SeeMeCNC Rostock printer and Cura version
> 4.8.0 can't be wrong, can they?

Hmm, puh... I feel your frustration...

Ghostkeeper, 2021-02-04:
If it's still generating wrong g-code when that setting is enabled, chances are that it also generated wrong g-code in Cura 4.6 and earlier. But that was really the whole problem with the setting. It wasn't doing what people asked of it, unless the correct changes were also made to extruder start and end g-codes. So we've removed it from the interface and the creators of the definition files are responsible to correctly set the extruder start and end g-codes for it to work out of the box. If you'd like to make improvements to the SeeMeCNC printer definitions, we'd love to incorporate them to make it work correctly for those users!