Ultimaker/Cura #9018: 4.8.0 Cura starts and crashes quietly after Window opens

jmueller42, 2020-12-25:
**Application version**
4.8.0 i also tried fresh reinstall of 4.71, but it also does not start anymore

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 x64 20H2
BuildNumber: 19042.685
Windows Feature Experience Pack : 120.2212.551.0

AMD 5900x
Nvidia 3080

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Just start the program

**Log file**
[stderr.log](https://github.com/Ultimaker/Cura/files/5742169/stderr.log)

nordlead2005, 2020-12-26:
This appears to be related to an update to the VC++ runtime. I uninstalled the 2015-2019 VC++ Runtime Redistributable and installed the 2015 version, and it works fine. However, my Logitech G HUB software and ASUS GPU Tweak II software need the latest version of the VC++ Redistributable. So, I'm stuck running either Cura, or my every day software, but not both.

Attached is the stderr.log file that generates when I use the latest C++ redistributables.

[stderr.log](https://github.com/Ultimaker/Cura/files/5743645/stderr.log)

jmueller42, 2020-12-27:
Uninstalling Asus GPU Tweak fixes it for me also. Thanks for the info.
For Asus GPU Tweak version 2.2.6.0 does not produce the problem. The problem with newer vc++ begins with version 2.2.7.0+ .

dgiambo1, 2020-12-28:
Thanks, this fixed my issue too. Uninstalled all vcredist, then installed the one in the Cura directory. For the record this was on 4.8.0. And mine was crashing on startup when hitting "Agree" for the data collection.

Ghostkeeper, 2020-12-28:
I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/8732. Seems like you've found a possible workaround though.

closed 2020-12-28 by Ghostkeeper