Ultimaker/Cura #9019: Allow half layer Zupport distance

printingotb, 2020-12-26:
**Is your feature request related to a problem? Please describe.**
Supports are difficult to tune for optimal results. We often get either _easy to remove_  or _smooth surfaces._ This setting would allow more room for optimization.

**Describe the solution you'd like**
Ideally we should be able to use any Z-distance but I think half of a layer would be enough (and easier to implement?).
This would only affect the top support layer.

**Affected users and/or printers**
Everyone

**Additional context**
As long as the halved support layer is printed before everything else, there would be no collisions.
[image]

hunt3dprints, 2020-12-26:
This would he a great help! The supports hate working with large layer heights having an option to change that would be awesome.

samtaylor3d, 2020-12-26:
Yeah I agree, this would be soooo handy with printing at larger layer heights with bigger nozzles.

When using layer height of 0.2 the z gap of 0.2 is perfect. When printing at 0.1 layer height 0.2mm z gap is also perfect 

But when printing at 0.4mm layer height the 0.4mm z gap is way too large and the features needing support either droop down or curl up. 0.2mm z gap would really help with this.  I know we can adjust the xy distance which helps but it isn't enough to get perfect supports with these larger layer heights. 

As said above ideally choosing any z gap at any layer height would be amazing but halving would be sufficient.

Ghostkeeper, 2020-12-28:
Duplicate of https://github.com/Ultimaker/Cura/issues/6105.

closed 2020-12-28 by Ghostkeeper

MariMakes, 2023-11-16:
Quick update on our side, you can now enter values that deviate from your layer height for the Support Z Distance. 
Can you please try it out and let us know if it resolves your issue?
You can download it here. https://github.com/Ultimaker/Cura/releases/tag/5.6.0-beta.1