Ultimaker/Cura #9023: [4.8.0] Coasting sometimes doesn't work regardless of 'Minimum Volume Before Coasting' setting

JohnTasto, 2020-12-26:
4.8.0
Windows 10
Heavily modified BIBO 2

As the title says, coasting is very inconsistent - sometimes it works on a layer, sometimes not, even if 'Minimum Volume Before Coasting' is set very low or even to 0.

Here is an example where on layer 100, there is a coast followed by a wipe:
[image]

On layer 101, there is only a wipe, but no coast for inner or outer walls:
[image]

I don't know why the Z seam corner abruptly changes either, another possibly related issue I often see. It is set to Sharpest Corner and Hide Seam.

This is not an isolated case; in fact every time I've tried enabling coasting this is about what I get.

Here is the project file:
[Coasting.zip](https://github.com/Ultimaker/Cura/files/5743588/Coasting.zip)

JohnTasto, 2020-12-26:
Here is a simpler example, this time on the same corner. Note the Z seam still bounces around all over the place regardless of 'Seam Corner Preference' setting, but here I found two adjacent layers with the seam on the same corner, one where coasting works and one where it does not. For this particular model, setting 'Seam Corner Preference' to 'None' results in the most layers with working coasting, but there are still a few where it does not.

Layer 36, coasting working:
[image]

Layer 37, coasting not working:
[image]

The project file:
[Coasting 2.zip](https://github.com/Ultimaker/Cura/files/5743648/Coasting.2.zip)

JohnTasto, 2020-12-26:
The last comment in #2973 suggests this might have something to do with combing, but neither of the four options for 'Combing Mode' appears to affect coasting.

Ghostkeeper, 2020-12-28:
The project files don't load in Cura because you've installed some sort of modded definition "bibent". Is there any printer definition that we'd need to install before your issue reproduces?

Thanks for finding that Compensate Wall Overlaps is (part of) the problem. That narrows it down quite a bit.

JohnTasto, 2020-12-28:
Ah, I wasn't sure what all was included in the save file. Here are my printer definition files (BIBent ~= bent BIBO). They're pretty basic; I only created them to change `nozzle_offsetting_for_disallowed_areas` to `false` and change `machine_nozzle_heat_up_speed` and `machine_nozzle_cool_down_speed`.

I also use some custom materials and a bunch of plugins - hopefully they won't cause problems.

[bibent.zip](https://github.com/Ultimaker/Cura/files/5748791/bibent.zip)

Ghostkeeper, 2020-12-30:
Indeed the project file contains some semblance of the definition file, but it's flattened due to how Uranium's data structure works; the inheritance in those files is transparent. It's not designed to work with modded Cura versions. Your file generates a bunch of warnings that it's trying to set Linear Advance values for me, but that's harmless.

I can reproduce the problem with your file now. It's weird that overlap compensation has that effect. The problem probably has something to do with how the line is broken up into pieces. Maybe something needs to change in the order, to apply coasting after the overlap compensation.
I'll propose this to the team. However I'm not sure how much chance there is that this'll be fixed since overlap compensation is removed in our [variable line width project](https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_alpha).

Ghostkeeper, 2021-01-12:
Indeed we've decided to not give this priority since Ultimaker/CuraEngine#1210 fixes it. Once that branch is merged, this issue will be closed.

Ghostkeeper, 2021-12-14:
https://github.com/Ultimaker/CuraEngine/pull/1210 is merged now, so we consider this issue fixed, by virtue of not doing the wall overlap compensation at all any more.

closed 2021-12-14 by Ghostkeeper