Ultimaker/Cura #9027: freeze

longlivekel, 2020-12-27:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-12-28:
Unless you fill in the template there is nothing that we can do for you.

longlivekel, 2021-01-11:
The issue was related to the latest software release for a mac. I
downgraded a version and resolved the issue.

longlivekel, 2021-01-11:
The freezing was caused by the latest version with m

closed 2021-01-11 by longlivekel