Ultimaker/Cura #9028: Lock orientation when using "Arrange All Models"

PatrickPaes, 2020-12-28:
Hi,
Recently I found a post by BradLevy in the Ultimaker Cura forum requesting the same thing (https://community.ultimaker.com/topic/35029-any-way-to-control-orientation-and-spacing-with-arrange-all-and-cura-48/) and thought it was worth posting here as well.

I run a small business that prints multiple small parts that requires to be printed in the same orientation in the build plate. Although I find the new Arrange All Models to be wonderful, it became less usable to me as I end up having to arrange models manually one by one which is very impractical when you have to arrange 50~100+ pieces.

A huge plus would be adjustable spacing.

This would help anyone that needs to print large batches of small items.

Hope to hear back from any of the devs.

Kindly,
Patrick.

Ghostkeeper, 2020-12-28:
We're aware of a problem due to dual extrusion, here: https://github.com/Ultimaker/Cura/issues/8597

Is there a reason you'd want to lock the orientation for just printing large batches of small (single-extrusion) items though? Why would you need these to keep their orientation?

PatrickPaes, 2020-12-28:
The products I manufacture must all have the same skin lines orientation,
so they can always match other lots and the way it was designed.

Those are small objects where the correct top surface skin lines
orientation plays an important role aesthetically. It is being sold in
lots, meaning it is very easy to spot when some pieces were printed in a
different orientation. Those objects are used in the film industry, so
professionals tend to notice more the little aesthetical details.

Let me know if it all makes sense.

Kindly,
Patrick

printingotb, 2021-03-13:
Another relevant reason is to get the Z-seam in the same position relative to each parts orientation.

d-portero, 2022-03-08:
I also would love to have the orientation locked when auto arranging. Some parts I print often have to be arranged in certain ways to facilitate better bridging. When auto arranging the models are rotated out of the optimal position

smartavionics, 2022-03-10:
I've just added this feature to my cura build. See https://github.com/smartavionics/Cura/releases if you want to try it out. Please open an issue at https://github.com/smartavionics/Cura rather than here if you have any problems/feedback.

smartavionics, 2022-03-10:
With rotations enabled (as before)...

[image]

With rotations disabled (now possible) ...

[image]

PatrickPaes, 2022-03-10:
This is amazing, good job Mark!
That's exactly what I was looking for. Works like a charm!

Cheers!

d-portero, 2022-03-10:
> I've just added this feature to my cura build. See https://github.com/smartavionics/Cura/releases if you want to try it out. Please open an issue at https://github.com/smartavionics/Cura rather than here if you have any problems/feedback.

I learned about your Cura build last week and have been so impressed. Can't believe I went this long without it

closed 2024-01-11 by MariMakes

PatrickPaes, 2024-01-13:
> Quick update from our side 👋
> We are cleaning up our open issues, so we can focus on the most requested
> and needed features and bug fixes from our community and from UltiMaker, I
> stumbled upon your issue about locking orientation when arranging models.
>
> I believe we introduced a feature that solves your problem last October in
> Cura 5.5. You can download the latest version of Cura to try it out for
> yourself.
> image.png (view on web)
>
> I'll close this ticket since there is a solution available.
> Thanks for your report. Together we can make Cura better 😊

That's awesome! Exactly what we needed.
In addition to that, if we could select spacing between them, that would be
great!
Just a suggestion.

Thank you Mariska!

Patrick Paes - (236) 863-9453
http://patrickpaes.com/