Ultimaker/Cura #9029: Arachne leaving wall gaps

epeckham85, 2020-12-28:
Installed today and printing a lithophane and some smaller parts. It looks like I am having some wall separation now on both printers. The new Cura brought in my old profiles so I used the same profiles only changing the z seam location due to know issue with shortest.
I am printing now with CURA 4.8 and I didn't see this issue. I also had Y-axis belt jump but I don't think the profile was the cause just a new printer belt stretch. 
Machine is an Ender 3 pro V2 that is all stock. 
[image]

Ghostkeeper, 2020-12-28:
Could you fill in the issue template for us?

epeckham85, 2020-12-29:
**Application version**
Cura Arachne Engine Alpha

**Platform**
Windows 10 x64.  Intel Core I7, Integrated IrisX GPU

**Printer**
Creality Ender 3

**Reproduction steps**
See above

**Screenshot(s)**
See above

**Actual results**
See above

**Expected results**
No gap

**Project file**
Attached
[CE3_NL-G&G Bratt Arachne.zip](https://github.com/Ultimaker/Cura/files/5749556/CE3_NL-G.G.Bratt.Arachne.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-12-30:
There is a 0.04mm gap because you've set the Flow setting to 90%. Please set the Flow to 100% if you want there to be no gap.

You can also consider using Skin Overlap if you're just running into [generic 3D printing issues](https://all3dp.com/1/common-3d-printing-problems-troubleshooting-3d-printer-issues/#gaps-between-infill-and-outer-wall). That's what it's there for.

closed 2020-12-30 by Ghostkeeper

epeckham85, 2020-12-30:
Sorry to waste your time. Thank you for looking into it. I will adjust.