Ultimaker/Cura #9030: [Arachne_engine_alpha] Connect infill lines results in unnecessary minimal extrusion blobs.

Cederb, 2020-12-28:
**Application version**
Arachne_Engine_Alpha 99.9.0

**Platform**
Windows 10 / Ryzen 1700x / GeForce GTX 1070

**Printer**
Ultimaker 2+

**Reproduction steps**
1. Open file 
2. Select 0.8mm nozzle
3. Select generic PLA, draft quality profile 
4. Select "Skirt" as build plate adhesion type
5. Check "Connect infill lines"
6. Slice

**Screenshot(s)**
This is with "connect infill lines" turned on, notice the row of minimal extrusion blobs in the corner.
[image]

This is with "connect infill lines" turned off
[image]

**Project file**
https://www.dropbox.com/s/qspkcuqn1o9cvkh/UM2_kapa-theeo-kompressor.3mf?dl=0

Ghostkeeper, 2020-12-28:
Hmm, quite recently we added something that would fill up thin strips of infill where it was missing due to the half-line-width offset that infill keeps when using Connect Infill Lines. That's probably the cause of this as well. Maybe the implementation was too simplistic.

Thanks for the report!

Ghostkeeper, 2021-01-12:
We've made a small fix that addresses this issue for the libArachne version.