Ultimaker/Cura #9032: Option to remove Cura icon in macOS toolbar

phrz, 2020-12-28:
**Is your feature request related to a problem? Please describe.**
I am only now noticing that Cura shows an icon in the macOS toolbar when minimized. It does not do anything when clicked (no dropdown or opening the app). With many apps on macOS vying for toolbar space, this potentially contributes to clutter.

**Describe the solution you'd like**
Please provide an easy option to remove this toolbar icon, e.g. a checkbox at the very top of Preferences > General > Interface, as opposed to somewhere more obscure. Even better: this functionality could be opt-in.

**Describe alternatives you've considered**
If this toolbar icon feature does not do anything, it could be removed for the time being, but I have no insight into its purpose.

**Affected users and/or printers**
All macOS users.

**Additional context**
[image]

Ghostkeeper, 2020-12-30:
The system tray icon is necessary to be able to show system tray messages, such as when your print is finished. At least, it's necessary in Qt. I don't know if this is required on all platforms, but it's the only method available to us.
We're using [QSystemTrayIcon.showMessage()](https://doc.qt.io/qt-5/qsystemtrayicon.html#showMessage) to do this.

Ghostkeeper, 2020-12-30:
It won't solve your problem, but I consider it a user experience problem that the icon does nothing. It should at least have some options to pause or abort an ongoing print or something.

phrz, 2020-12-30:
@Ghostkeeper I am no Qt expert, but did a little research on Qt, and it seems like the tight coupling of System Tray Icons with Messages is a relic of e.g. Windows XP (which is still the OS in the example image in the [documentation](https://doc.qt.io/qt-5/qsystemtrayicon.html#details)). 
[image]
Although macOS (and Windows 10) allows applications to post notifications without regard for a toolbar icon, Qt seems to need one to have the other. Yet KDE's Notifications add-on for Qt ([repo](https://invent.kde.org/frameworks/knotifications), [docs](https://api.kde.org/frameworks/knotifications/html/classKNotification.html)) apparently overcomes this design issue and lets programs post rich notifications to the OS's Notifications panel. It's certainly not mandatory to use this, but if the system tray icon is only there out of obligation, there may be some value in switching from `QSystemTrayIcon` to `KNotification`.

Ghostkeeper, 2021-01-08:
It's a bit problematic to use that since we'd need to build Python bindings for that then. We currently use Qt through PyQt and PyQt is quite a complex project in scope. It doesn't include external Qt add-ons of course.

jake-nz, 2021-01-18:
For those of us using Cura as a slicer only, there's no need to system tray messages. Perhaps a setting to opt-out of both messages and the icon would be a good idea? It would definitely be fine for me.

Ghostkeeper, 2022-02-26:
Duplicate of https://github.com/Ultimaker/Cura/issues/2687

closed 2022-02-26 by Ghostkeeper