Ultimaker/Cura #9034: Settings numbers don't align with boxes

medicriley, 2020-12-29:
[image]

if the drag and drop worked you will see the numbers in settings don't align with the boxes. This is with any settings selection and new ones. I use Windows 10, I am using the current build of Cura as of 12/28/2020. I tried deleting and reinstalling the program, I did not remove the settings

medicriley, 2020-12-29:
Worth noting it still works fine.

Ghostkeeper, 2020-12-30:
Weird! I've never seen anything like this.

> I am using the current build of Cura as of 12/28/2020

I think you mean the 4.8 release then, right?

I'm not able to reproduce this issue. I'll ask around when my colleagues are back from their vacations if they have any idea what could cause this (I'm the only programmer working right now).

medicriley, 2020-12-30:
It was originally on every setting I had. I created a new setting profile
and it all just went away! So I am good. Hope all is well, thank you.

Michael

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Ghostkeeper, 2021-01-04:
Maybe it's similar to the [Intel driver bug](https://github.com/Ultimaker/Cura/issues/2068) we encountered a long time ago, but it only happens with text? Or maybe it's some leftover system-themed margins causing this.

None of our other developers or testers has seen this issue. We'll keep it open for now to see if other people also have the issue, and if so what the commonalities between your computers are where the issue reproduces.

closed 2024-12-08 by github-actions[bot]