Ultimaker/Cura #9038: Cura Froze after material export

DonBower, 2020-12-29:
**Application version**
4.8.0

**Platform**
macOS Catalina 10.15.7

**Printer**
Ender 

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
[image]

**Actual results**
Cura Froze

**Expected results**
Not Freeze

**Project file**
None

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5751761/cura.log)

**Additional information**
Had to Force Quit

Ghostkeeper, 2021-01-06:
I've not been able to reproduce this issue so far on my computer. It's properly exporting the file and not freezing.

The fact that the pop-up appears with a success message shows that the file was properly exported completely, and the file handle closed. Writing that file should have nothing to do any more with the freeze. But I don't see why a simple pop-up would cause Cura to freeze. Maybe there is another pop-up underneath that has the actual focus or something?

DonBower, 2021-01-09:
After rebooting my Mac, I too, am unable to replicate this issue.
We should just close this i think.

closed 2021-01-11 by nallath