Ultimaker/Cura #9039: [4.8.0]  Crash when i make a new material

Nexus77777, 2020-12-29:
**Application version**
4.8.0

**Platform**
Win 10/64

**Printer**
Custom

**Reproduction steps**
1. Ctrl+K
2. click "Create"
3. Change any value and it instantly crashes, every time.
4. Just installed the new version 4.8.0, in the previous beta version i used i had the same issue.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
New project (empty)

Thread 0x00001a3c (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000034a4 (most recent call first):
  File "threading.py", line 297 in wait
  File "threading.py", line 549 in wait
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00003b98 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1262 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002fb0 (most recent call first):
  File "zeroconf\__init__.py", line 1183 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00004ac4 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000045ac (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000021d8 (most recent call first):
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
[... 85 lines cut ...]
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

**Log file**
Traceback (most recent call last):
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\AutoSave.py", line 61, in _onTimeout
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 723, in saveSettings
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\cura\Settings\CuraContainerRegistry.py", line 487, in saveDirtyContainers
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 604, in saveContainer
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\XmlMaterialProfile\XmlMaterialProfile.py", line 126, in setDirty
    if base_file is not None and base_file != self.getId() and not registry.isReadOnly(base_file):
  File "X:\4.8-exe\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 321, in isReadOnly
  File "C:\Program Files\Ultimaker Cura 4.8.0\plugins\LocalContainerProvider\LocalContainerProvider.py", line 188, in isReadOnly
    file_path = self._id_to_path[container_id]  # If KeyError: We don't know this ID.
KeyError: 'custom_material #8'

**Additional information**
(Extra information relevant to the issue.)

clb92, 2021-01-03:
I have the same problem.

Nexus77777, 2021-01-03:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAGAAAABgCAYAAADimHc4AAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAAA7SURBVHic7cExAQAAAMKg9U9tCF+gAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACA1wCQYAABH00o1QAAAABJRU5ErkJggg==###24x24:true###][Tracking-Bild][https://github.com/notifications/beacon/ASJRPJYSSZPL57724HYB47LSX7NQVA5CNFSM4VNQDOI2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFTVG3HA.gif]

I found a workaround in the meantime. If you duplicate a existing profile you then can change it without crashing, also rename it.

Ghostkeeper, 2021-01-06:
Do you happen to have the Start-up Optimizer plug-in installed from the Marketplace? It has a symptom like this.

forenbuddy, 2021-01-14:
Same here. Workaround not working and "duplicate a existing profile" also not working. Crashes as well.

clb92, 2021-01-14:
> 
> 
> Do you happen to have the Start-up Optimizer plug-in installed from the Marketplace? It has a symptom like this.

Disabling Start-up Optimizer did indeed fix it for me, it seems. Thanks!

fieldOfView, 2021-01-14:
Could you test if [this version](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/StartOptimiser/BENphMMk-ed8BKmjOVbw-aXf5sPngjEaJWlcXf-cOD-Y/StartOptimiser-v7.0.0-2021-01-11T10%3A12%3A57Z.curapackage?Expires=1925998017&GoogleAccessId=cura-packages-api-production%40stardust-193112.iam.gserviceaccount.com&Signature=LVzzFGanbmBpJHF0OWXtf6Hsdkm0FRY15%2BdcefAOn8nwg%2FflSja%2B8kAGUsiyvboYJtb4oSNfI0qrpBN2u%2FbDynVJvVbspHXegxeeFEjYbrB9vmyfCQS7FHdlmL1OWbM6Ob4GZaMI7b2%2BiiRNV5Fs0XQbHTeXvzAWnfz8h%2FZsIGer8QJuyLpqgyd%2FXh8mOTcD2r3wLC8dlisTmcvuyvo0KrsksrCyJ%2FBBcdbiNuJHkiMnNNy1k4mtjzdSaIRXVY9kO3ET6GhwADrs%2Fjj%2B7onG8tVADedM6U6q%2FTOLD%2BI0z8NaoWtGcPVZQTTG%2BdJmUYqlhnpxTOdB9IH6i2AMwzzRNQ%3D%3D) of the plugin fixes the issue for you? Download the file and drop it onto the buildplate in Cura as if you were opening a 3d model.

clb92, 2021-01-14:
> Could you test if [this [...]

No crashes for me when creating or duplicating materials with this version.

tsipizic, 2021-01-14:
Seems to be a problem with the Arch Linux build since I have it too and the workaround does not help but if I use the appImage it works normally.

Ghostkeeper, 2021-01-25:
> Disabling Start-up Optimizer did indeed fix it for me, it seems. Thanks!

> No crashes for me when creating or duplicating materials with this version.

Then I'll consider this to be a bug in that plug-in. I've recently approved fieldOfView's new version so updating the plug-in should fix it for everyone.

> Seems to be a problem with the Arch Linux build since I have it too and the workaround does not help but if I use the appImage it works normally.

Then the maintainer of the Arch repository will have made a mistake. One common mistake is that Cura requires Qt 5.10 at the moment and may crash with newer versions of Qt. We consider this out of scope. We only maintain the AppImage for Linux.

closed 2021-01-25 by Ghostkeeper