Ultimaker/Cura #9040: [Feature Request] Create Purge Line Option

KenLucke, 2020-12-29:
**Is your feature request related to a problem? Please describe.**
I know that I can create a purge line with my printer’s start-code.  In fact, I have the following start code:

```
; Machine Setup
M201 X500.00 Y500.00 Z100.00 E5000.00 ; Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ; Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ; Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ; Setup Jerk
M220 S100 ; Reset Feedrate
M221 S100 ; Reset Flowrate
; End Machine Setup

G28 ; Home

M420 S1 Z5 ; use stored mesh, bed level on, fade height 5mm

; Purge Line Code
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up
G1 X10.1 Y30.0 Z0.28 F5000.0 ; move away from clip
G1 X50.0 Y30.0 Z0.28 F5000.0 ; move around clip
G1 X50.0 Y5.0 Z0.28 F5000.0 ; Move to start position
G1 X250.0 Y5.0 Z0.28 F1500.0 E11 ; Draw the first line
G1 X250.0 Y5.3 Z0.28 F5000.0 ; Move back a little
G1 X50.0 Y5.3 Z0.28 F1500.0 E23 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up
; End Purge Line Code
```

However, there are times and models with which I do not want to create a purge line, but yet I want to leave the acceleration/jerk/etc settings in place, which requires me to either temporarily edit my start gcode, or manually edit the file after slicing.

**Describe the solution you'd like**
I’d like to see a “Create Purge Line” option, which would enable it to be easily turned on and off for each model. There should be options for Placement (Front, Back, Left or Right), Distance From Edge, and Length (the line would be centered [Distance from Edge] along whichever edge chosen).  

**Describe alternatives you've considered**
Continuing to manually edit either the machine setup each time (RPITA), creating another machine profile without the purge line, which requires creating all the other profiles associated with a machine, or manually editing the file after slicing.

Ghostkeeper, 2021-01-06:
I know that this was being considered at some point when the Ultimaker 3 printer was under development. But it was then decided that this should be implemented in the firmware and instead, the `G280` command was created and the "Enable Prime Blob" setting was added to Cura.

From experience with that prime blob, there are Ultimaker profiles in Cura which have it enabled and profiles which have it disabled. One reason to disable it is that the material may let go of the build plate more easily with certain materials, which makes the purged material be dragged along with the nozzle. But otherwise it's quite a good replacement for a skirt, since it prints faster than a skirt.

So I'd be in favour of adding such a setting to Cura, for printers that don't have such a prime blob.

However... This is not something that would really apply to Ultimaker printers. It only applies to some of Ultimaker's competitors. I don't think we're going to get any time or priority to implement this then. For that reason, to be realistic in the communication, I'll defer this issue. It's a real issue and a good feature, but it's not going to be implemented soon. The most realistic chance is that someone else develops it for us through a pull request. Sorry!

KenLucke, 2021-01-06:
Grrrr..... I can understand, in  one way, but on the other hand, it's frustrating to be told it's a good feature and not yet not going to be implemented because it's not relevant just to some Ultimaker printers makes one wonder why their promo phrase "The World's Most Advanced 3D Software at the Heart of Your Workflow" doesn't read "The World's Most Advanced 3D Software at the Heart of Your Workflow As Long As You Buy And Use Our Printers."  They promote it for all printers, but only support their own?

nallath, 2021-01-11:
> They promote it for all printers, but only support their own?

Well, yes. We have to make choices with what to build / fix. So things that benefit our own machines will get priority over those that don't. Part of that is because making software is expensive so we have to ensure that our cashflow remains steady & ongoing. Another part is that we simply don't have any non-um printers to test it on.

the-grizzman, 2021-06-01:
I agree, this feature would be very useful. I'm tired of jamming a putty knife under the nozzle at the beginning of each print to clear out the runaways until it starts printing cleanly. Once it gets going, it prints great, but I'm going to have to go back to Voxelab slicer for now - it has a pre-extrusion setting that works great, I don't have to babysit the start of every print.

BloodyRain2k, 2021-07-22:
Wouldn't this be something that could be band-aided with a post processing script?
There would just be a setting needed that adds ANY marker to the GCode so the script knows it has to do it's thing, or not.

Ghostkeeper, 2021-07-29:
You could enable the Prime Blob setting, which puts a G280 command when the nozzle first needs priming (also correctly for dual extrusion). That is a marker that you're looking for. The setting is hidden by default though unless using the Griffin g-code flavour, so you'd have to make it visible in the printer definition.

Then you could have a search-and-replace post-processing script replace `G280` with something like `G0 X10 Y10\nG1 X110 E3.5`.

closed 2023-07-13 by github-actions[bot]