Ultimaker/Cura #9044: Open File attempt Crashes Cura 4.8

dearmonds, 2020-12-30:
**Application version**
Cura 4.8

**Platform**
Dell XPS 8700 Intel i7, Windows 10 Pro 64bit

**Printer**
Ender-3 Pro

**Reproduction steps**
1. Open Cura 4.8
2. Select any Open File Icon or Menu Item

**Screenshot(s)**
None available

**Actual results**
Program Closes (presumed to crash)

**Expected results**
Program should allow selection of a file to load, but closes shortly after file manager view opens.

**Project file**
N/A

**Log file**
None available.

**Additional information**
Tried uninstalling and reinstalling, but did not help.

dearmonds, 2021-01-01:
Seems to be fixed by uninstalling Dell Recovery and Backup application as mentioned in other threads.   I would like not to have to do that.

Ghostkeeper, 2021-01-08:
If it's not creating a log file, most likely none of our own code was even ran but the process was killed before CPython properly started.

There's one thing we could try to guess at what's going wrong. You could open a command line (Win+R, then type `cmd`). Then run Cura from there: `"C:\Program Files\Ultimaker Cura 4.8\cura.exe"`. This provides a little bit more information since there is some output in stderr from the Qt framework, and also some print statements from before it loads the logger plug-in.

I wonder how a back-up application could prevent Cura from running though. It doesn't delete the AppData directory or anything I'd hope...

closed 2021-01-22 by no-response[bot]