Ultimaker/Cura #9045: Cura v4.8 can not import a 3mf file

omache55, 2020-12-30:
That 3mf file were saved by nTopolopy Element software.
Then tried to import those file and nothing showed  it on Cura display.
Is there a way to convert old 3mf format to your new 3mf format?
That nTopolopy Element software is old and no longer around on online

bremco, 2021-01-02:
If they adhere to the 3mf standard, we should at least be able to open the file (there isn't really a 'new' or 'old' version in the sense that, as far as I know, if there is a polygonal model inside, we should be able to load it).

Perhaps we can have a look inside if you attach the file? (You may have to zip it to make github accept the format.) It's very hard for us to guess at what the developers of discontinued software thought a valid 3mf looked like.

omache55, 2021-01-03:
first attached file was wrong one - second one is 3mf format

bremco, 2021-01-03:
I'm sorry, I don't see any attached files. Please be aware that email attachments don't show up on github.

omache55, 2021-01-03:
I upgraded Chrome today because of its GL error recently which might cause
the image function not to work.
So, I redo it if I have any luck...I got it from my folder (nothing do with
any of GitHub stuffs)

Ghostkeeper, 2021-01-08:
E-mail attachments don't get uploaded to Github. Probably because too many people had images in their signatures and stuff. Please visit the actual web page here and upload it to a comment (by dragging the file in): https://github.com/Ultimaker/Cura/issues/9045

closed 2021-01-22 by no-response[bot]