Ultimaker/Cura #9046: Cura plugin development - help needed?

peta, 2020-12-31:
Hello everybody,

I am just getting started with Cura plugin development and thus decided to use an existing plugin and add some small improvements to it which I find quite useful. Thus, It's the `PostProcessignPlugin` I am talking about and for this I did just copy it 1:1, renamed it and started hacking.

So far I was able to implement the functionality but now, on the very last mile, I am faced by a PyQt-specific issue to which I have absolutely no idea how to continue. Looking on the screenshot below, you see that now every active script has its own checkbox which can either be toggled manually one after another, or by using the enable/disable all buttons (both, in the dialog and inside the context menu of the according secondary button in the `postProcessingSaveAreaButton` pane).

Now the problem: as long as I do not toggle one of these checkboxes manually, using the enable/disable all buttons work as expected. But as soon as I did manually toggle a checkbox, this one will not get properly redrawn when using the mentioned bulk-mode buttons. Their internal model state gets properly updated, so it seems to me as if for some reason the checkbox control doesn't get updated anymore in the the Qt view. However, as soon as I add another script, the list gets redrawn and this time the affected checkbox get redrawn. Very strange.

Do you have any idea? 

[image]

Here is the repository: https://github.com/peta/Cura-PostProcessingPlugin

Thanks & all the best
Peter

P.S. Don't be irritated that I did copy the plugin and prepended its naming with "My " , as said it's just 1) for learning purposes and not meant for distribution in official plugin channel and 2) that my custom version can co-exist parallel with the original plugin.

bremco, 2021-01-02:
We made a few plugin examples especially for learning purposes, although I don't know how up-to date they are: https://github.com/Ultimaker/Cura/wiki/Plugin-Directory (Also don't worry about prefixing stuff with 'My'; it signals exactly what you want it to do, to the point that it's basically the coders equivalent of those 'L' signs on cars/trucks/buses people learn how to drive in.)

That said, the problem at hand: I think that the 'checked' properties binding has been overwritten with a hard 'true' or 'false' the moment the user clicked on the checkbox. (See for example [this stackoverflow question](https://stackoverflow.com/questions/23860270/binding-checkbox-checked-property-with-a-c-object-q-property).) 

As for when it _does_ work when the list is changed: I think it's forced to recreate the checkboxes, thus restoring the bindings, if the list of strings (the `QStringList` PyQt-property `scriptList`) is altered in any way just beyond the index. (Say, the order of them: You can see this if you randomize the list given back by the `scriptList` PyQt-property, the update works, although the indices are of course wrong then.)

In any case, personally, I'd rewrite it anyway so that the model the list relies on isn't just the (list of) strings any more, but some QObject or somesuch, that includes the 'is_active' state. That saves you from mucking around with a separate indexed list of booleans, which will also solve this problem as a bonus. (Well, hopefully. Qt can be a finicky beast. It all makes sense in the end but getting there is another thing.)

P.S. Since this is a (variant of a) bundled plugin, you could also try forking Cura, altering the PostProcessingScript itself in a branch (or on main if that's the only PR you'll have going on), and then making a pull request. Eventually I mean, when it's all working and the code shines again.

closed 2021-01-08 by Ghostkeeper

Ghostkeeper, 2021-01-12:
> We made a few plugin examples especially for learning purposes, although I don't know how up-to date they are:

I still have the plug-ins loaded and test them once in a while. Especially the tool example since it's easy to reach.