Ultimaker/Cura #9049: Cura crashes on project saving and material creation.

a123qwertz567, 2020-12-31:
**Application version**
4.7.1 (from manjaro repos); 4.8 (Appimage from Cura website); 4.8 (git version installed via arch AUR)

**Platform**
Linux-5.9.16-1-Manjaro-x86_64-with-glibc2.32
Qt version: 5.15.2
PyTw version: 5.12.2
OpenGL Version: 4.6 (Core Profile) Mesa 20.3.1
OpenGL Vendor: AMD
OpenGl-Renderer: AMD Radeon (TM) RX 480 Graphics

**Printer**
I tested multiple printers:
- Custom FFF printer (from Ultimaker)
- Sapphire Pro (With custom mesh and config, placed inside curas config folder)

**Reproduction steps**
- Save project file does not work.
- I have no clear way to reproduce it.
- I think it can be related to the nozzle size parameter and the diameter of the filament set in the printer settings

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
When I create a new printer and new material and try to save the project cura crashes.

**Expected results**
No crash.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
**_Error traceback:_**
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 1060, in event
    return super().event(event)
  File "/usr/lib/python3.9/site-packages/UM/Qt/QtApplication.py", line 464, in event
    event._function_event.call()
  File "/usr/lib/python3.9/site-packages/UM/Event.py", line 218, in call
    self._function(*self._args, **self._kwargs)
  File "/usr/lib/python3.9/site-packages/cura/Utils/Threading.py", line 34, in _handle_call
    ico.result = func(*args, **kwargs)
  File "/usr/lib/cura/plugins/3MFWriter/ThreeMFWorkspaceWriter.py", line 57, in write
    self._writeContainerToArchive(container, archive)
  File "/usr/lib/cura/plugins/3MFWriter/ThreeMFWorkspaceWriter.py", line 150, in _writeContainerToArchive
    serialized_data = container.serialize(ignored_metadata_keys = ignore_keys)
  File "/usr/lib/cura/plugins/XmlMaterialProfile/XmlMaterialProfile.py", line 154, in serialize
    builder.start("metadata") # type: ignore
TypeError: start expected 2 arguments, got 1

**_Log:_**
Thread 0x00007f186ffff640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 2432 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1726 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f1890a79640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/threading.py", line 574 in wait
  File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f189127a640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1476 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f1891a7b640 (most recent call first):
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1350 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f189227c640 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f1892a7d640 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f18937fe640 (most recent call first):
  File "/usr/lib/python3.9/site-packages/UM/Backend/Backend.py", line 165 in _storeStderrToLogThread
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f1893fff640 (most recent call first):
  File "/usr/lib/python3.9/site-packages/UM/Backend/Backend.py", line 153 in _storeOutputToLogThread
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f18cbfff640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 574 in wait
  File "/usr/lib/python3.9/site-packages/cura/Utils/Threading.py", line 39 in _call_on_qt_thread_wrapper
  File "/usr/lib/python3.9/site-packages/UM/FileHandler/WriteFileJob.py", line 76 in run
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 132 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f18d0ff9640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f18d17fa640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f18d1ffb640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Current thread 0x00007f190bc14740 (most recent call first):
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 393 in _logInfoWidget
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 182 in _createDialog
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/bin/cura", line 166 in exceptHook
  File "/usr/lib/python3.9/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 844 in run
  File "/bin/cura", line 234 in <module>

**Additional information**
I updated my system today and switch from a 5.7 kernel to a 5.9 kernel. With the update came lot of python updates, too.
I reseted cura completely. I can export gcode and slice just saving is not possible.

a123qwertz567, 2020-12-31:
Workaround for now:
I can load a old profile with all settings and slice in that setting. But I cannot save stuff....

a123qwertz567, 2021-01-01:
Updated the issue. Cura 4.6 as an Appimage seems to work for now. Saving a project makes it hang/lag for some seconds but it works for now.

delaosa, 2021-01-05:
Same crash with AUR package (cura-4.8.0-1) in Arch however 4.8.0 Appimage is working well.

Ghostkeeper, 2021-01-08:
Cura currently requires Python 3.5. Perhaps this is one of the places where Python's interface changed to not be backwards compatible?

The line in question is here:

https://github.com/Ultimaker/Cura/blob/9562dc01f47bc1b3a368fa80563677a914bb09a0/plugins/XmlMaterialProfile/XmlMaterialProfile.py#L154

That `#type: ignore` makes me a bit suspicious that the argument list was also seen as wrong on some computers but correct in the Python version that Cura runs with.

aghriss, 2021-01-12:
The problem is in the file "/usr/lib/cura/plugins/XmlMaterialProfile/XmlMaterialProfile.py"
You just have to replace all calls
`builder.start(X)`
with
`builder.start(X,{})`

`# Copyright (c) 2019 Ultimaker B.V.
# Cura is released under the terms of the LGPLv3 or higher.

import copy
import io
import json #To parse the product-to-id mapping file.
import os.path #To find the product-to-id mapping.
from typing import Any, Dict, List, Optional, Tuple, cast, Set, Union
import xml.etree.ElementTree as ET

from UM.PluginRegistry import PluginRegistry
from UM.Resources import Resources
from UM.Logger import Logger
import UM.Dictionary
from UM.Settings.InstanceContainer import InstanceContainer
from UM.Settings.ContainerRegistry import ContainerRegistry
from UM.ConfigurationErrorMessage import ConfigurationErrorMessage

from cura.CuraApplication import CuraApplication
from cura.Machines.VariantType import VariantType

try:
    from .XmlMaterialValidator import XmlMaterialValidator
except (ImportError, SystemError):
    import XmlMaterialValidator  # type: ignore  # This fixes the tests not being able to import.

class XmlMaterialProfile(InstanceContainer):
    """Handles serializing and deserializing material containers from an XML file"""

    CurrentFdmMaterialVersion = "1.3"
    Version = 1

    def __init__(self, container_id, *args, **kwargs):
        super().__init__(container_id, *args, **kwargs)
        self._inherited_files = []

    @staticmethod
    def xmlVersionToSettingVersion(xml_version: str) -> int:
        """Translates the version number in the XML files to the setting_version metadata entry.

        Since the two may increment independently we need a way to say which
        versions of the XML specification are compatible with our setting data
        version numbers.

        :param xml_version: The version number found in an XML file.
        :return: The corresponding setting_version.
        """

        if xml_version == "1.3":
            return CuraApplication.SettingVersion
        return 0  # Older than 1.3.

    def getInheritedFiles(self):
        return self._inherited_files

    def setMetaDataEntry(self, key, value, apply_to_all = True):
        """set the meta data for all machine / variant combinations

        The "apply_to_all" flag indicates whether this piece of metadata should be applied to all material containers
        or just this specific container.
        For example, when you change the material name, you want to apply it to all its derived containers, but for
        some specific settings, they should only be applied to a machine/variant-specific container.

        Overridden from InstanceContainer
[... 1080 lines cut ...]
    }

def _indent(elem, level = 0):
    """Helper function for pretty-printing XML because ETree is stupid"""

    i = "\n" + level * "  "
    if len(elem):
        if not elem.text or not elem.text.strip():
            elem.text = i + "  "
        if not elem.tail or not elem.tail.strip():
            elem.tail = i
        for elem in elem:
            _indent(elem, level + 1)
        if not elem.tail or not elem.tail.strip():
            elem.tail = i
    else:
        if level and (not elem.tail or not elem.tail.strip()):
            elem.tail = i

# The namespace is prepended to the tag name but between {}.
# We are only interested in the actual tag name, so discard everything
# before the last }
def _tag_without_namespace(element):
    return element.tag[element.tag.rfind("}") + 1:]
`

delaosa, 2021-01-12:
Thanks @aghriss, it works for me.

c24w, 2021-01-17:
> The problem is in the file "/usr/lib/cura/plugins/XmlMaterialProfile/XmlMaterialProfile.py"
> You just have to replace all calls:
> `builder.start(X)`
> by
> `builder.start(X,{})`
> There are a few such calls between line 154 and 196

Thanks @aghriss; works for me.  Here's a command to do it in one go:

`sed -i 's/\(builder\.start([^),]\+\))/\1, {})/' XmlMaterialProfile.py`.

Use at your own risk, of course :grimacing:

a123qwertz567, 2021-01-17:
Thank you for the solution. I will not try it if it works I am fine for now. The whole settings, profile, filament management is a mess from my perspective. I won't waste time to dial in setting in something that can break tomorrow....

Ghostkeeper, 2021-01-25:
This has recently been fixed for us through a pull request: https://github.com/Ultimaker/Cura/pull/9105

The changes suggested by @c24w and applied by @lorenzph are backwards compatible to Python 3.5. So it apparently fixes it for Python 3.9, and doesn't break anything for the main Cura distribution. So we've merged them in.
Thanks for the debugging, guys :)

closed 2021-01-25 by Ghostkeeper