Ultimaker/Cura #9052: Cura timing way off

SCRATMAN06, 2020-12-31:
**Application version**
 4.8.0

**Platform**
Windows 10

**Printer**
Anycubic chiron

**Reproduction steps**
1. Slice model
2. Print model

**Actual results**
example- 2 hours

**Expected results**
Example- Cura says 1 hour 20 minutes

Asterchades, 2021-01-01:
Cura's timing is based on the information it has. If any settings which can be adjusted at the firmware level are different - typically going to be acceleration/jerk and maximum feed rates - then the estimates won't match.

For example, the default Chiron profile uses the base acceleration listing of a completely impractical 9,000m/s² for X/Y. From what I've been able to find out with a brief search, the Chiron defaults to a rather sedate 350mm/s² in firmware on those axes. Because of this disparity, any corner in the model will put the estimate off simply by virtue of the calculation being done using a significantly wrong value. The more corners, the less accurate.

You can either give Cura the correct acceleration values (a plugin such as "Printer Settings" can do this, or you can manually edit the config) or use the acceleration and jerk override values within Cura ("Speed" sub-section) such that the actual print and the estimate are using matching data. It still won't be exact (it cannot account for things like minimum segment times in Marlin or delays in processing), but it should be much closer.

Ghostkeeper, 2021-01-11:
So this is an issue of having inaccurate machine settings. You could try downloading the Printer Settings plug-in from the Marketplace and adjusting the maximum acceleration / feedrate / jerk machine settings that become visible then. I suggest looking up the correct values for them in your printer's configuration source code (`Configuration.h` for Marlin).

Cura's time estimation is based on the state of Marlin when Ultimaker branched off its Marlin version about 5 years ago. Nothing has changed in this regard on Ultimaker's side since then. If newer Marlin versions have new features that affect the printing time that will not be taken into account yet. These new Marlin features can be implemented in Cura, but since they are not in Ultimaker printers we can't really build them as we don't have anything to verify them by nor any benefit to Ultimaker's users.

Other than those new features, the estimate is as far as we know very exact, with just a few caveats:

* Cura doesn't round to (micro)step sizes.
* Cura assumes that the printer's CPU is ideal. That is, it doesn't slow down the print head because the CPU needs to process too many commands in a short amount of time.
* Cura assumes that the printer's clock speed is ideal. In practice we've seen a few % variation in clock speed which affects the timings of the pulses sent to the stepper motors.

Ultimaker holds that unless there is buffer underrun, the time estimate should be within 5% of the actual time. We test that with our own printers.

closed 2021-12-30 by Ghostkeeper