Ultimaker/Cura #9053: [4.8.0] Weird layer added in middle of cube with 0% infill

JohanAR, 2020-12-31:
**Application version**
4.8.0 appimage

**Platform**
Xubuntu 20.04.1

**Printer**
Creality Ender-3

**Reproduction steps**
1. Create primitive cube in FreeCAD
2. Create mesh and save as STL
3. Import in Cura
4. Set Standard Quality and 0% infill
5. Slice and preview results

**Screenshot(s)**

[image]

**Actual results**
Layers 67 - 75 have weird horizontal triangles

**Expected results**
Layers 67 - 75 should be empty (except outer walls)

**Project file**

[cura_cube.zip](https://github.com/Ultimaker/Cura/files/5758333/cura_cube.zip)

**Log file**

[cura.log](https://github.com/Ultimaker/Cura/files/5758338/cura.log)

bremco, 2021-01-02:
I'm not sure what's going on here, but I did find a workaround: Set the 'Slicing Tolerance' to 'Inclusive'.

Ghostkeeper, 2021-01-11:
Inclusive will fix gaps like this, yeah. Still shouldn't occur though.

The width of the gap happens to be exactly up to the border of the triangle that bisects one of the sides. It's like it's ignoring the two vertices in the corners on that layer only. Very weird.

Ghostkeeper, 2021-01-11:
The model itself is just a simple cube and seems to be perfectly fine.

GregValiant, 2024-11-19:
I haven't seen this bug in the latest versions.  Changes somewhere else may have fixed this.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

closed 2024-11-20 by GregValiant