Ultimaker/Cura #9055: [4.8] GUI freezing randomly every few secs

lordgix, 2021-01-01:
**Application version**
4.8 (tryed also 4.7.1, same behaviour)

**Platform**
Windows 10, i-9700K, 32GB RAM, RTX-2080 SUPER

**Printer**
Ender 3 (v2)

**Reproduction steps**
1. Fresh install Cura 4.8
2. Open Cura and follow initial wizard to set up printer (I might get some slowness/unresponsiveness at this step already)
3. Go through Cura menus, or profile settings (don't even need to load a file)
4. Cura freezes randomly, sometimes displaying "Unresponsive" in the top bar

**Screenshot(s)**
[image]
https://user-images.githubusercontent.com/63344121/103440850-46423600-4c49-11eb-8318-fc8e3818990b.mp4

**Actual results**
Cura freezes, and comes back to live after some seconds most of the times.
Sometimes, the app hangs forever and I have to exit through task manager.

**Expected results**
Shouldn't freeze.

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5759042/cura.log)

**Additional information**
Working with a 2-screen setup, one G-sync compatible, the other not. Issue happening on both screens.
NVIDIA drivers updated to latest available
Tryed tweaking NVIDIA and Windows settings to force use of GPU but nothing worked

lordgix, 2021-01-03:
I installed 4.6 and seems much more stable. It crashes from time to time requiring to close it via task manager but at least it doesn't freeze every few moments. What could be the difference with 4.7 and 4.8?

lordgix, 2021-01-10:
Been working with 4.6 without issues for a week. Gave 4.8 another try and it froze again, but I believe I found the root cause for the issue: there were some unreacheable mapped network drives. Once drives are reachable, Cura 4.8 stops freezing.

closed 2021-01-10 by lordgix

Ghostkeeper, 2021-01-11:
If those network drives are mapped to the `X:` driveletter, this is a duplicate of https://github.com/Ultimaker/Cura/issues/1602 . There may be more information in the other ticket. It's a very weird bug.