Ultimaker/Cura #9056: 4.6.1/4.8.0 Bridge Skin/Sparse Infill doesn't activate Bridging if Infill is 0%

sg000, 2021-01-01:
**Application version**
4.6.1 and 4.8.0

**Platform**
Windows 10 64 Bit

**Printer**
Custom

**Reproduction steps**
Take any model. Activate Bridging.
Set infill to 0% and the Bridge Sparse Infill Max Density to Anything greater than 1
Slice
*notice no bridging.
Go back and set infill to 1% (making sure the sparse infill is still greater than 1
Slice
*notice we now have bridging, but also 1% of infill (which I did not desire)

I tried changing other settings like Support Threshold, Wall length but until infill is not 0%, bridge skin just doesn't activate.
This is not helpful when trying to create an empty cavity without any infill

Ghostkeeper, 2021-01-11:
Indeed, I could reproduce this issue. You removed the "Project file" entry from the template so it's always a gamble, but I could find your case: [infill_density_0_not_bridging.zip](https://github.com/Ultimaker/Cura/files/5796535/infill_density_0_not_bridging.zip)

0% density is a bit of a special case in CuraEngine since it skips the entire infill generation then. That's probably what's causing this issue.

sg000, 2021-01-11:
> Indeed, I could reproduce this issue. You removed the "Project file" entry from the template so it's always a gamble, but I could find your case: [...]

I'll remember not to delete that.

Also, found a workaround in a way. It does accept non-integer for the infill %, so setting to 0.1% provides desired outcome for my model anyways.

Ghostkeeper, 2021-01-25:
We'll recognise this issue as a real problem, but it doesn't get any priority from us now. We'll defer it.

The main reason for this decision is that it's a bit of an edge case. While there's good reasons for the user to look specifically to this edge case (when printing with 0% infill, bridging is more important), it's a pretty rare combination of settings and involves the experimental bridging feature as well.

When we're going to push to bring those bridging settings to a more stable state, we'll probably take a look at this again. But not for now, unless a fix is contributed to us externally.

hoffr, 2022-05-11:
this happens when setting top/bottom layers to 99 (effectively 'disabling' infill(?)). this is what i do when i want 100 percent infill, i just use all skin because i typically use an extra infill wall for lower density infills, but i don't want it when doing 100% infill because it messes with the patterns, so i just increase skin since the infill line would just be another setting to toggle and keep track of

sg000, 2023-06-11:
> Hi 👋, We are cleaning our list of issues to improve our focus. This bug seems to be older than a year, which is at least three major Cura releases ago. It also received the label Deferred indicating [...]

As far as I know this still isn't fixed. I've continued to use 0.1% infill on all new releases, but I've not specifically tested otherwise. I recognize its an edge case (user wants no infill, but does want bridging). Workaround is still to use 0.1%, but new users won't figure this out easily.

MariMakes, 2023-07-10:
Hey @sg000,

I'm a fresh set of eyes, reporting for duty 🖖 
Sorry, it took us a while to get back to you 😞 

I can confirm that this issue is still present in Cura 5.4. I also recognize that the workaround isn't ideal because it results in infill being printed where you don't want it 🤔 
[image]

I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

MariMakes, 2023-07-18:
Quick update from our side 👋

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10823.

Thanks for the report! 👍

MariMakes, 2023-10-11:
This is resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-11 by MariMakes