Ultimaker/Cura #9057: [Arachne_engine_alpha] Suddent closes after eject memory card

hugoalves6, 2021-01-01:
**Application version**
Cura Arachne Engine Alpha

**Platform**
SO: Windows 10 
Board: B450 Aorus Elite
CPU: Ryzen 7 2700x
GPU: Radeon RX 5700XT
RAM: 4x G. Skill Ripjaw V 4GB CL17 3733mhz
Ecra 1: LG MP59G IPS 75hz
Ecra 2: Asus VG24V VA 144hz Curvo
Ecra 3: LG 24MK600M IPS 75hz
Disco 1: SSD NVMe Samsung 970 EVO Plus 250GB
Disco 2: SSD NVMe Crucial P1 1TB
Disco 3: SSD Crucial MX500 500GB
Disco 4: HDD Seagate Barracuda 1TB

**Printer**
Custom : Anet A8 & Anet A8 Plus Single and also Triple extruder

**Reproduction steps**
1. After saving gcode to memory card, clicked eject, and the program closed

**Screenshot(s)**
Don't have

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5759546/cura.log)

**Additional information**
It doesn't appends all the times.

nhopfen, 2021-01-03:
I also have the same problem. It seems to happen every time in my case. I just give it a few seconds and pull out my sd without pressing eject to circumvent this.  Otherwise I only have good things to say about the new engine, especially for surface  details such as letters and for variable width/sloping walls, this is a total game changer.

Prickly-Pear, 2021-01-07:
This problem also occurs on MacOSX (v10.15.7). After saving gcode to memory card, clicking the eject button makes the program close unexpectedly.

HubvX, 2021-01-07:
Yes, I tried and it happens every time.
Now testing resulting gcode which seems to be ok.
32 Mb gcode file.

hugoalves6, 2021-01-09:
For me is more like in 10 saves, 1 crashes the program

Ghostkeeper, 2021-01-11:
We've had this report before, although we can't reproduce the issue yet so far. Developers, refer to CURA-7947.

ksihota, 2021-02-04:
I have noticed that Cura (arachne_alpha) closes when you select the popup dialog to eject the SD card and it also closes when you select the popup dialog to reload a changed file (an STL altered outside of Cura while the model is active in Cura.)
Cura will not close if you right click on the model and reload it using the context menu, only if you reload using the popup. Maybe the issues are related to the popup dialog interaction?

konskarm, 2021-04-08:
We just released the [4.9 beta](https://github.com/Ultimaker/Cura/releases/tag/4.9-beta). Can you guys give it a try to see if the issue persists? So far we haven't been able to reproduce it in the 4.9 beta release.

hugoalves6, 2021-04-11:
> We just released the [4.9 beta](https://github.com/Ultimaker/Cura/releases/tag/4.9-beta). Can you guys give it a try to see if the issue persists? So far we haven't been able to reproduce it in the 4.9 beta release.

Yep, it still crashes the program, i open it like some 20 times today, but only now at saving to the sd card the program crashed.
With aracne engine never append to me.

ksihota, 2021-04-12:
I just did a quick check of Arachne_engine_beta and of 4.9 beta.
Arachne_engine_beta still closes when selecting the popup dialog for ejecting the SD card and using the popup dialog for updating a changed model. This does not occur if you ignore the popup and use eject SD from Windows or use the right mouse button menu (context menu) for reloading all models.
I just installed 4.9 Beta and tested it as well and this issue does not appear to happen with it.

hugoalves6, 2021-04-12:
> I just did a quick check of Arachne_engine_beta and of 4.9 beta.
> Arachne_engine_beta still closes when selecting the popup dialog for ejecting the SD card and using the popup dialog for updating a [...]

well for me is quite the oposite, this problem doesn't append on arachne anymore but on 4.9 beta it does

ksihota, 2021-04-12:
I tried arachne_engine_beta again with the SD card save and then eject using the popup dialog. My first try appeared to work without closing down Cura but then I tried again using a different SD card and Cura closed down. To check if it was the SD card I repeated the same procedure using different cards but keeping everything else consistent and Cura closed down each time., so I assume it was not the SD card that made the difference. I have included the Log lines related to the shut down and the one time it did not shut down below:
**Arachne_engine_beta sliced and selected the Save to SD**

2021-04-12 09:52:21,499 - DEBUG - [MainThread] RemovableDriveOutputDevice.RemovableDriveOutputDevice.requestWrite [85]: Writing to D:/ZAP_Connector Block.gcode
2021-04-12 09:52:21,516 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._parseJWT [100]: There was no auth data or access token
2021-04-12 09:52:21,780 - DEBUG - [Thread-7] UM.FileHandler.WriteFileJob.run [80]: Writing file took 0.25597071647644043 seconds
2021-04-12 09:52:22,938 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [284]: SliceInfo sent successfully
2021-04-12 09:52:23,871 - INFO - [MainThread] RemovableDriveOutputDevice.RemovableDrivePlugin.ejectDevice [44]: Attempting to eject the device
2021-04-12 09:52:23,894 - INFO - [MainThread] RemovableDriveOutputDevice.RemovableDrivePlugin.ejectDevice [51]: Succesfully ejected the device

**Successful Ejection of SD card (Cura remains running)**

2021-04-12 09:52:23,899 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Program Files/Ultimaker Cura Arachne_engine_beta 99.9.1/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
2021-04-12 09:52:56,566 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura Arachne_engine_beta 99.9.1\resources\shaders\toolhandle.shader]...
2021-04-12 09:52:59,977 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura Arachne_engine_beta 99.9.1\resources\shaders\toolhandle.shader]...
2021-04-12 09:53:02,398 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: GroupedOp.(#=2) RotateOp.(node=<CuraSceneNode object: 'Connector Block.stl'>,rot.=Q<0.7071067811865475,0.0,-0.0,w=0.7071067811865476>) GravityOp.(node=<CuraSceneNode object: 'Connector Block.stl'>), took 21ms
2021-04-12 09:53:06,078 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend._createSnapshot [256]: Creating thumbnail image (just before slice)...
2021-04-12 09:53:06,082 - DEBUG - [MainThread] UM.Scene.Camera._preferencesChanged [255]: Changing perspective mode to perspective.
2021-04-12 09:53:06,085 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura Arachne_engine_beta 99.9.1\resources\shaders\overhang.shader]...
2021-04-12 09:53:06,112 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura Arachne_engine_beta 99.9.1\resources\shaders\transparent_object.shader]...
2021-04-12 09:53:06,131 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [C:\Program Files\Ultimaker Cura Arachne_engine_beta 99.9.1\resources\shaders\striped.shader]...
2021-04-12 09:53:06,230 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [271]: Starting to slice...

**Resliced and selected Save to SD**

2021-04-12 09:53:08,298 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [737]: Slicing took 2.0679430961608887 seconds
2021-04-12 09:53:08,299 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [738]: Number of models per buildplate: {0: 1}
2021-04-12 09:53:08,299 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [752]: See if there is more to slice...
2021-04-12 09:53:08,303 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Closing
2021-04-12 09:53:08,305 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2021-04-12 09:53:08,307 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
2021-04-12 09:53:08,308 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [925]: Backend quit with return code 0. Resetting process and socket.
2021-04-12 09:53:22,082 - DEBUG - [MainThread] RemovableDriveOutputDevice.RemovableDriveOutputDevice.requestWrite [85]: Writing to D:/ZAP_Connector Block.gcode
2021-04-12 09:53:22,097 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService._parseJWT [100]: There was no auth data or access token
2021-04-12 09:53:22,406 - DEBUG - [Thread-1] UM.FileHandler.WriteFileJob.run [80]: Writing file took 0.297804594039917 seconds
2021-04-12 09:53:24,763 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [284]: SliceInfo sent successfully
2021-04-12 09:53:26,199 - INFO - [MainThread] RemovableDriveOutputDevice.RemovableDrivePlugin.ejectDevice [44]: Attempting to eject the device
2021-04-12 09:53:26,221 - INFO - [MainThread] RemovableDriveOutputDevice.RemovableDrivePlugin.ejectDevice [51]: Succesfully ejected the device
2021-04-12 09:53:26,240 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [475]: Shutting down cura
2021-04-12 09:53:26,297 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [341]: Attempting to kill the engine process
2021-04-12 09:53:26,302 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [411]: file:///C:/Program Files/Ultimaker Cura Arachne_engine_beta 99.9.1/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null

**Cura closes down after selecting Eject SD card from popup dialog**

Ghostkeeper, 2021-04-13:
Some Ultimaker colleagues were also able to reproduce this issue on Windows. We've confirmed that the issue was fixed by the upgrade to Python 3.8 / Qt 5.15. So this is not an issue any more in Cura 4.9, and will not be an issue in any next releases we'll make of Arachne either.

Thanks for the report! We wouldn't have found this since none of the testers or developers had this issue (it seems to happen only on some computers). Luckily it got fixed as a side-effect of another fix we made.

closed 2021-04-13 by Ghostkeeper