Ultimaker/Cura #9058: 4.8 Post Processing Filament change does not properly return to proper Z

Grey-Lancaster, 2021-01-02:
**Application version**
4.8

**Platform**
Windows 10 saving to Micro Sd Card

**Printer**
Creallity Ender 3

**Reproduction steps**
1. Use post processing change filament at layer

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
I have problems with 4.8 change at layer vs 4.5 change at layer. Both show M600 in the code but they behave differently. 4.5 returns to 0,0 far left bed retracted. 4.8 returns to the middle of the bed 110? with the bed all the way back. That is not actually an issue I just ponder what is happening.  If I hit change filament on the LCD it goes to 0,0.

My issue is maybe it is not locking the stepper motor?  Seems like 8 times out of 10 I manage to lower the "z" and when it returns to print the nozzle is burning a hole in my print.

**Expected results**
Z should go back to the proper value for the next layer

**Project file**
[45.zip](https://github.com/Ultimaker/Cura/files/5759577/45.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2021-01-11:
Note that the 4.8 project file doesn't produce any filament change command since the post-processing script sets the layer trigger to layer 100, but the print doesn't have 100 layers.

Ghostkeeper, 2021-01-11:
I'm not seeing a whole lot of difference in the way that Cura behaves here. The `M600` command is exactly the same. And the surrounding g-code has different coordinates (due to the new arranger and the new simplification algorithm most likely) but the structure is the same.

This is a snippet of the 4.5 output (my comments):

```
G1 X65.243 Y84.759 E3554.07951 ;Last normal print move.
G0 F9000 X65.249 Y84.595
G1 F2700 E3549.07951
;MESH:NONMESH
G0 F300 X65.249 Y84.595 Z2 ;Moving to Z=2 in advance of the next layer.
G0 F9000 X65.336 Y84.938 ;Bunch of travel moves start here to get to the starting position.
G0 X66.349 Y85.44
G0 X67.954 Y119.422
G0 X97.06 Y122.915
G0 X97.055 Y126.091
G0 X110.147 Y143.423
G0 X107.712 Y143.423
G0 X109.256 Y146.06
G0 X109.256 Y146.113
;TIME_ELAPSED:4438.244924
M600 X0.00 Y0.00 ; Generated by FilamentChange plugin ;Exactly the same as in 4.8.
;LAYER:9
;TYPE:WALL-INNER
;MESH:Object 1
G1 F2700 E3554.07951 ;Unretraction.
G1 F1500 X109.16 Y146.213 E3554.08642 ;Starting printing again...
G1 X109.122 Y146.262 E3554.08952
```

And this is a snippet from 4.8:

```
G1 F1500 X66.961 Y81.103 E3397.81727 ;Last normal extrusion move.
G0 F9000 X67.018 Y81.339
G1 F2700 E3392.81727
;MESH:NONMESH
G0 F300 X67.018 Y81.339 Z2 ;Move to Z=2 in advance of the next layer!
G0 F9000 X68.509 Y83.502 ;Bunch of travel moves to get to the starting position of the next layer.
G0 X69.216 Y83.8
G0 X69.688 Y85.09
G0 X69.688 Y86.069
G0 X94.864 Y91.965
G0 X96.03 Y92.744
G0 X96.81 Y93.91
G0 X97.071 Y95.221
G0 X97.06 Y119.422
G0 X110.809 Y136.431
G0 X110.809 Y134.996
G0 X112.424 Y135.438
G0 X112.415 Y137.877
G0 X112.971 Y138.874
G0 X116.059 Y138.881
G0 X116.441 Y145.09
G0 X116.618 Y145.692
;TIME_ELAPSED:6328.215216
M600 X0.00 Y0.00 ; Generated by FilamentChange plugin ;Exactly the same as in 4.5.
;LAYER:9
;TYPE:WALL-INNER
;MESH:Gingerbread_Kit_A.stl
G1 F2700 E3397.81727 ;Unretract.
G1 F1500 X116.658 Y146.06 E3397.82958 ;Continue printing.
```

In both cases, the next Z command is at the end of the next layer, some 6000 commands later (`Z2.2`).

Ghostkeeper, 2021-01-11:
The two look identical to me. The printer should behave the same. What would we need to improve in the g-code output to make it function correctly?

Grey-Lancaster, 2021-01-15:
I was hoping it was a known issue and someone would jump in with a fix.  I will have to dbl check the code I posted.

Outside the box.

- So it used to work fine. Could it be possible my stepper motors are "weak" and do not lock during a change?
- On my Ender 3 for the majority of my prints it would not affect the print if the head returned to 0,0,0.  Could there be a post processing script (return home) that would go to 0,0,0 and then raise up and go to where it should be.  In case x,y,z got moved during a filament change?

Thanks,
Grey

fvrmr, 2021-01-22:
Hi @Grey-Lancaster 

> * So it used to work fine. Could it be possible my stepper motors are "weak" and do not lock during a change?

It is possible that your stepper motor receive not enough power. If you want to know how to measure it, please let me know. 

> * On my Ender 3 for the majority of my prints it would not affect the print if the head returned to 0,0,0.  Could there be a post processing script (return home) that would go to 0,0,0 and then raise up and go to where it should be.  In case x,y,z got moved during a filament change?

The x and y are already homing:
> M600 X0.00 Y0.00 ; Generated by FilamentChange plugin

The Z is not homing, because if you will home Z the changes are that the object will be destroyed by the x and y. 😉 
But I will discuss this with the team. Keep you posted!

Grey-Lancaster, 2021-01-22:
See that is what I do not understand.  The code says M600 x0.0 and y0.0 but it goes to X 0.00 and Y 110.0
If I hit filament change on the LCD it goes to 0,0

But I just noticed that filament change on the LCD does not lock any of the stepper motors.  A Pause print does.

I was wrong 4.5 does also go to x0 and Y 110.  I am old, but I swear it used to go to ),) and I never had an issue with changes like this.

Perhaps I need to upgrade the motherboard?  This is an Ender 3 by SainSmart

On the reset Z concept.  My print very seldom fill the bed and most of the color changes are done at less than layer 15.  Doing luggage tags, signs, dog tags, that kind of thing.

Thanks again for looking at this.
Grey

Ghostkeeper, 2021-01-25:
> But I just noticed that filament change on the LCD does not lock any of the stepper motors. A Pause print does.

This could be your answer then. Some printers kill power to the motors while the print is paused, to reduce energy usage and wear on the steppers. While power is killed, it's easy to move the gantry by hand and this would then destroy any knowledge of where the print head is until it's homed again. Changing filament is one way to "pause" it so perhaps the issue is indeed that it's losing track of its Z coordinate. The only real way to fix that is to home Z again after the filament change then.

I'd consider this to be a bug in your firmware, rather than Cura. Normally the pause operation doesn't need homing again, and homing can also cause a layer shift because the endstops don't always respond in precisely the same location. I don't think upgrading the motherboard would have any effect. Changing the firmware would. If it's keeping the power on for normal pause, then it should certainly keep the power on for a filament change pause.

As a workaround, you can add a post-processing script that inserts a home command after every M600. This is done with a Search-and-Replace post-processing script with regex enabled. You'd need to search for the following string:

    (M600.*\n)

And replace that by:

    \1G28\n

closed 2021-01-25 by Ghostkeeper

Grey-Lancaster, 2021-01-28:
Thanks
I needed to add a GO Z after the G28 but that worked fine.
I managed to do the TH3 firmware and it appears to lock all the motors.
Thanks for your insight and attention to my issue.
Grey