Ultimaker/Cura #9062: camera rotation

hickokchris, 2021-01-02:
**Is your feature request related to a problem? Please describe.**
The problem: rotating the camera always summons the context menu. The real problem: Ultimaker Cura often interprets my attempts to rotate the camera as deleting all objects on the build mat. 

**Describe the solution you'd like**
Hotkey+left click rotates the camera. This is a standard user experience practice for 3d programs across all industries and disciplines.

**Describe alternatives you've considered**
I've exhausted the menu options and researched this issue with no resolution.

**Affected users and/or printers**
Affected users: all

**Additional context**
(Add any other context or screenshots about the feature request here.)

Ghostkeeper, 2021-01-11:
> Hotkey+left click rotates the camera. This is a standard user experience practice for 3d programs across all industries and disciplines.

I don't know of any 3D application that uses a hotkey together with left-click to rotate the camera. In general there are no real standards here though. Blender rotates the camera with the middle mouse button. OpenSCAD uses the left mouse button. In BricsCAD it's even Ctrl + right mouse button.

We've had plans for a long time to allow the user to select from a number of camera control configurations. We're imagining a drop-down with options like "SolidWorks", "Blender", "AutoCAD", "FreeCAD", "OpenSCAD", to allow the user to select the camera controls they are used to. This has been a desire for a long time though and we haven't gotten around to it yet. I'll ask around what the status of that is in our planning.

nallath, 2021-01-12:
We already had a ticket for this. 

Developers, see CURA-6841

daanta-real, 2024-07-11:
Not resolved

adamkerz, 2025-01-22:
I've looked at https://github.com/Ultimaker/Uranium/blob/main/plugins/Tools/CameraTool/CameraTool.py and the logic seems fairly straightforward to customise based on a different `_navigation_style`. A little refactoring of a couple of methods would make it easier to decouple the inputs from the actions so that different navigation styles can be added easily. I'll start with Blender as I know it.

@nallath, @Ghostkeeper if I raise a PR on that repo, is it likely to get attention/is there any resistance to adding this functionality?
(I can't see the JIRA ticket)