Ultimaker/Cura #9063: Add Snapmaker 2 printers to Cura

maihde, 2021-01-02:
**Is your feature request related to a problem? Please describe.**
The Snapmaker 2 printers are not available in Cura by default.  It would be very handy to have Cura come pre-installed with Snapmaker 2 profiles.

**Describe the solution you'd like**
I'd like Snapmaker 2 to be available as a printer in the "Add a Printer" dialog.

**Describe alternatives you've considered**
Many Snapmaker users pass around Cura profiles and instructions that can be used for setting up custom printers through the user-interface.  This can be error-prone and doesn't integrate well with Cura's ability to have materials override temperature/retraction settings.

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)
Snapmaker 2 users will benefit.

**Additional context**
I've already started creating the necessary printer configurations here:

https://github.com/maihde/snapmaker_cura

I have been using them for about a month without issues.  You may copy these into Cura's repository or, if preferred, I can make a pull request.

fieldOfView, 2021-01-02:
> You may copy these into Cura's repository or, if preferred, I can make a pull request.

A pull request is the only way.

maihde, 2021-01-02:
@fieldOfView thanks, I will prepare a PR!

schulzch, 2021-01-11:
For reference:

- Luban's G-Code [addMachineStartGcode()](https://github.com/Snapmaker/Luban/blob/e4255076866f6d64213d5b9d07253bd8ac10176c/src/app/flux/printing/DefinitionManager.js#L200) and [addMachineEndGcode()](https://github.com/Snapmaker/Luban/blob/e4255076866f6d64213d5b9d07253bd8ac10176c/src/app/flux/printing/DefinitionManager.js#L249)
- Luban's [machine images](https://github.com/Snapmaker/Luban/tree/master/src/app/images/machine)
- [Snapmaker Logo](https://github.com/Snapmaker/Luban/blob/master/src/app/images/snapmaker-logo-512x128.png)

You should be able to use those since Luban's AGPLv3 is compatible with Cura's LGPLv3. Hope it helps.

maihde, 2021-01-11:
@schulzch thanks.  I had already pulled the relevant start/end GCode along with the settings into [this](https://github.com/maihde/snapmaker_cura/blob/main/definitions/snapmaker2.def.json) and [this](https://github.com/maihde/snapmaker_cura/blob/main/definitions/snapmaker2_A350.def.json) along with the various default Luban [quality settings](https://github.com/maihde/snapmaker_cura/tree/main/quality/snapmaker2).  I've been using these profiles with my Cura for about a month now and things have worked well.

I made a few decisions based off other examples or my preference.  For example, I exclude a large set of the Cura built-in materials in the same way `creality_base` does.  I personally prefer this but I don't know what the preferred Cura approach is.

I hadn't thought of including the machine images/logo, thanks for pointing those out!  I'll include those with the PR.

Ghostkeeper, 2021-01-11:
> I personally prefer this but I don't know what the preferred Cura approach is.

Our "official" stance on this is that materials should not be excluded unless they are tested and found to be incompatible with the printer. It is rather annoying and anti-competitive if Cura were to explicitly forbid your favourite filament.

I'm going to close this issue, as being something we can't fix (we've closed such issues before to manage expectations). If you add the PR and it works well, we'll of course be happy to merge it in though.
And feel free to keep asking questions, either here or in your PR.

closed 2021-01-11 by Ghostkeeper

maihde, 2021-01-11:
Thanks @Ghostkeeper .  That makes sense...I will remove the exclusions because I have no reason to believe any of them to be incompatible at this time.

highpowerxh, 2021-12-07:
@maihde I think you should use the layer 0 temperature for the bed and hotend to honor the settings for initial layer. I will send a PR shortly