Ultimaker/Cura #9064: Skirt is over the model on wireframe slicing

diogorolo, 2021-01-02:
**Application version**
4.6.2 and 4.8.0 at least

**Platform**
Windows 10

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Import the attached project: 
[CE3PRO_WP_Leonardo.zip](https://github.com/Ultimaker/Cura/files/5760435/CE3PRO_WP_Leonardo.zip)

2. Set wireprint in the Experimental options

**Screenshot(s)**
Model in Cura (no preview is available for wireprinting so I can't show that)
[image]

**Actual results**
I didn't take a photo of the actual print before cutting cleaning, so the issue is not visible as printed. But using a GCode visualizer we can see the issue:
[image]
The skirt is printed inside the area of the model.

**Expected results**
The skirt should not overlay the model.

**Project file**
[CE3PRO_WP_Leonardo.zip](https://github.com/Ultimaker/Cura/files/5760435/CE3PRO_WP_Leonardo.zip)

**Log file**
2021-01-02 15:23:42,343 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] [WARNING] Mesh has overlapping faces!
^^ Not sure if relevant

**Additional information**
-

GregValiant, 2021-01-08:
I have seen that when the bottom of the model isn't exactly flat.  Try moving the model down into the build plate by 1mm and see if the problem goes away.  An alternative is to view the model from the underside and see if there are any red areas that need support.  If the model isn't quite flat you could increase the skirt distance.

GregValiant, 2021-01-09:
Moving it down didn't work.  The skirt is fine in regular mode and does go under the D in wire mode.  Changing the skirt distance had no effect.

Ghostkeeper, 2021-01-12:
Indeed the skirt in Wireframe is hard-coded to be a skirt (regardless of the Adhesion Type setting) and in fact hard-coded to be a single loop, always 5mm away from the model: 

https://github.com/Ultimaker/CuraEngine/blob/746bca02cebb7a7a3890e5886fabe356a563684c/src/Wireframe2gcode.cpp#L639

Many things are hard-coded in there and not properly using the settings. I'm afraid they won't get proper attention since Wireframe is more on its way out sort of as an unsuccessful experiment. It takes too much effort to maintain.

I also see the issue that's causing this bug. The skirt is doing an approximation of a convex hull there by making a big offset (100mm) and then an inset of the same size. This is what the skirt always does; it's what produces the shrinkwrap-like behaviour of the skirt. However the offset is normally with round joints. Here it's using a miter. This causes more jagged edges, and in this case also cuts off the protrusion from the ascender of the `d`.

I've fixed it here: https://github.com/Ultimaker/CuraEngine/commit/9ead8ee0574db21ab39d17dbf2b0ce70d1120041

closed 2021-01-12 by Ghostkeeper